iNavFlight/inav #4927: Mamba405 - xm+ Sbus not working

distractedfpv, 2019-07-11:
mamba405 and xm+ not working

i have nothing changed on the hardware, in betaflight 4.0 everything works as intended. 
when i flash one of the two inav firmwares for the fc it shows random things on the receiver pad.
Port tab is correct, config tab is on sbus. 
so far i tried all combinations of:

firmware 
serialrx_inverted 
serialrx_halfduplex

the only time i got it to work was when i used the hardware uninverted sbus, but this pin is basically not useable because its not ment to be used and way too small, also as said in betaflight it works just fine with serialrx_inverted and serialrx_halfduplex set to OFF

distractedfpv, 2019-07-11:
The Solution is here!
https://github.com/iNavFlight/inav/issues/4262#issuecomment-464488277

if i flash those .hex files sbus works like intended. Please get them in the firmwareflasher

digitalentity, 2019-07-14:
Should be fixed by #4936. Firmware for testing is available in that PR

closed 2019-07-14 by digitalentity