iNavFlight/inav #4931: Servos not recentering in AIRMODE

robthomson, 2019-07-13:
I have an odd issue on some models when airmode is enabled.

Essentially with ACRO, centring the radio stick swings the servo back to centre.  As expected.

In AIRMODE, if you put the stick on the radio to center; the servo stays at the position preceding.

So.. full right.  And release.  Results in the control surfaces staying at full right.

Disable air mode.. and problem goes away.

Below is a dump of my config.

I have reflashing etc.. same issue.

Bit confused really!

Rob

--
# resources

# mixer
mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix 0 1 0 -100 0 -1
smix 1 2 1 100 0 -1
smix 2 3 1 100 0 -1
smix 3 2 2 100 0 -1
smix 4 3 2 -100 0 -1

# servo
servo 2 1000 2000 1500 -100
servo 3 1000 2000 1500 60
servo 4 1000 2000 1500 60

# logic

# feature
feature -AIRMODE
feature MOTOR_STOP
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial

# aux
aux 0 0 0 1650 2100
aux 1 20 0 900 2100
aux 2 9 1 1450 1550
aux 3 8 1 900 1075
aux 4 35 1 1800 1900

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 26 2 V
osd_layout 0 1 1 12 V
osd_layout 0 3 8 6 H
osd_layout 0 4 8 6 H
osd_layout 0 7 13 11 V
osd_layout 0 9 1 2 H
osd_layout 0 11 3 6 H
osd_layout 0 12 11 12 V
osd_layout 0 13 24 10 V
osd_layout 0 14 25 1 V
osd_layout 0 15 24 11 V
osd_layout 0 20 1 2 V
osd_layout 0 21 1 1 V
osd_layout 0 22 1 8 H
osd_layout 0 23 1 11 V
osd_layout 0 27 23 9 V
osd_layout 0 28 23 12 V
osd_layout 0 30 1 6 V
osd_layout 0 32 13 3 H
osd_layout 0 45 0 0 V
osd_layout 0 97 2 8 H

# master
set looptime = 4000
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU9250
set acczero_x = 45
set acczero_y = 56
set accgain_x = 4089
set accgain_y = 4097
set accgain_z = 4033
set mag_hardware = NONE
set magzero_x = 49
set magzero_y = 818
set magzero_z = 420
set baro_hardware = BMP280
set pitot_hardware = ADC
set rssi_channel = 8
set motor_pwm_rate = 50
set model_preview_type = 8
set 3d_deadband_low = 1425
set 3d_neutral = 1475
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set deadband = 10
set yaw_deadband = 10
set mc_airmode_type = THROTTLE_THRESHOLD
set nav_extra_arming_safety = OFF
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 5000
set nav_fw_cruise_thr = 1600
set nav_fw_min_thr = 1400
set nav_fw_max_thr = 2000
set nav_fw_loiter_radius = 4000
set osd_hud_radar_range_min = 1
set vtx_band = 3
set vtx_channel = 8
set vtx_freq = 5945

# profile
profile 1

set fw_p_pitch = 10
set fw_i_pitch = 40
set fw_ff_pitch = 15
set fw_p_roll = 10
set fw_i_roll = 30
set fw_ff_roll = 15
set fw_p_yaw = 16
set roll_rate = 60

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 310
set battery_capacity = 5000
--

DzikuVx, 2019-07-13:
> with a confidence of 0.70. Please mark this comment with 👍 or 👎 to give

This is how it's expected to work on a workbench

digitalentity, 2019-07-13:
This is how it works on the ground.

robthomson, 2019-07-13:
Uhm..  for my and others benefit.... why?

How does holding the old position achieve anything?
And why does it not work like this in 'air mode' on other models of mine?

I feel I have a gap in my knowledge somewhere that needs addressing.

DzikuVx, 2019-07-13:
Feedback loop is not working when air is not moving through control
surfaces so error accumulates over time.

digitalentity, 2019-07-13:
First thing you need to keep in mind - in any stabilized mode there is **no direct connection** between your stick and the servos. It's your FC that flies the plane trying to respect your commands.

In AIRMODE you command rate of rotation and FC executes a feedback loop. Since your plane sits on the bench, when you deflect the stick FC tries to execute your command. However, FC sees no reaction to your command and tries to push servos more, hoping to get the plane moving.

Then, when you release the stick - you command "no rotation". FC happily executes. Since FC sees "no rotation" even though servos are not centered, there is no reason from firmware perspective to center them back. Pilot wants zero rotation. Plane is not rotating. Who cares about servo position.

robthomson, 2019-07-13:
OK.. So essentially in theory it should not be a problem.

But  seems very worrying to launch a model with control surfaces out of wack. 

Guess solution is to do all control checks while not armed. 

Feels to me like 'air mode' should only kick in one throttle gone up and you are away.

Will rest and see if change in my launnch procedure sill results in a flyable model. 

What worries me with this is that on my FPV acrowot, I attempted to fly with this scenario and nearly wrote the model off.   I had to manual recentre controls my applying opposite control!

Guess wil need to put on a switch so I can pass through if a problem.

robthomson, 2019-07-13:
> First thing you need to keep in mind - in any stabilized mode there is **no direct connection** between your stick and the servos. It's your FC that flies the plane trying to respect your commands. [...]

This is very helpfull info.

I have just bench tested on the ground and observed this in action.  If I apply control; then  physically move the model, the controls centre as expected.

Thanks for the explanation.

closed 2019-07-14 by DzikuVx