iNavFlight/inav #4934: Runcam Swift 3 Micro control not working

stephenaa, 2019-07-13:
## Current Behavior

The Runcam Swift3 can be set in "Remote mode" by stick control (Throttle middle, yaw right.
However, no other stick actions seem to result in any reaction from the camera. 

The following reference: https://support.runcam.com/hc/en-us/articles/360010541514-How-to-do-wiring-and-setup-for-your-remote-control-version-Racer-Micro-Eagle-and-Micro-Swift-3-

## Steps to Reproduce

1. Throttle/yaw and elevator/aileron in neutral
2. Yaw right, observe "REMOTE MODE" from RunCam device
3. Yaw right again
4. Yaw left

## Expected behavior

Step 3 should result in the Runcam device's menu to appear - it does not.
Step 4 should result in the Runcam device to exit "REMOTE MODE" - it does not

## Suggested solution(s)

## Additional context

I am able to debug, build, modify code - even hookup oscilloscope - I just need some context and pointers in the right direction.
---

- FC Board name and vendor: "Wing FC 10 DOF" from Banggood
- INAV version string: INAV/FF_35_LIGHTNING 2.2.1 Jul 3

kreidler, 2019-08-07:
Even if this issue is a bit older:
- Input on throttle in receiver tab is 1500?
- Have you hooked up to UART3 of the WING FC? Seems to be inverted according BG description. Try to use UART5

joao-ufjf, 2021-03-26:
Similar issue here. 
I have a MAMBA F405 MK2, the brown wired on RX6, the green wired on TX6.
UART6 on RunCam Protocol.

I changed the Camera to UART Control already, but nothing shows up when do left stick right.
The message when it sstarts still showing to "long press up..."

Any help?