iNavFlight/inav #4935: OMNINXT F7: Gyro0 aligment wrong. Gyro2 not working. RX2 pin not working.

alex-vasilyev-myob-com, 2019-07-14:
## Current Behavior
There are few issues with latest 2.2.1 firmware for OMNINXT F7 FC. 

1. Firmware settings for alignment of onboard gyro (gyro_to_use=0) is incorrect. Model does not move as expected. If board alignment changed than model moves ok but quad insta crash because of ACC vs GYRO alignment is wrong.

Works as expected on Betaflight 4.x. In BF 4 gyro 0 alignment is 0 and gyro 1 alignment is CW270.

2. If gyro_to_use set to 1 and second gyro is MPU600 board just hang and cannot boot. Require re-flash. 
If gyro hardware changed to ICM20602 then board boot but gyro not detected properly (MPU6500 instread of ICM) and accelerometer not detected at all. Changing acc hardware manually does not help. 

3. RX2 pin is not working. I have CRSF RX connected and it works fine in BF 4.0. After I flash iNav 2.2 receiver is no longer works, but telemetry still does. It seems RX2 pin is not working? Moving RX to UART3 helps, but just curious about pin.

## Steps to Reproduce
1. Flash iNav 2.2.1 and observe model in setup window. Orientation is wrong.
2. Use CLI to set gyro_to_use=1, observe no accel or hang.
3. Connect CRSF to UART2, no RX detected while TLM working.

## Expected behavior
Should work as in BF 4.x.

---
- FC Board name and vendor: Airbot OMNINXT F7
- INAV version string: 
# version
# INAV/OMNIBUSF7NXT 2.2.1 Jul  3 2019 / 22:26:41 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

digitalentity, 2019-07-14:
1. Which board exactly do you use? Original one from Airbot or compatible clones?
2. Was it working before (in 2.1 or 2.0)?

digitalentity, 2019-07-14:
Tested on my OMNIBUS NXT board.

Both gyro0 (MPU6000) and gyro1 (ICM20602) work as expected. ICM20602 gyro is handled by MPU6500 driver, so this is also working as expected. After changing gyro you also need to switch accelerometer as it's only detected once, also working as intended.

RX (IBUS, don't have CRSF at hand) on UART2 and UART3 work. No issue here.

The only issue that I can confirm - this board only supports MPU6500-compatible gyro as external gyro. MPU6000 is not expected to work at the moment.

To troubleshoot further - provide the board model and link to the shop and your config.

alex-vasilyev-myob-com, 2019-07-14:
> 1. Which board exactly do you use? Original one from Airbot or compatible clones?

I have two boards which I bought directly from airbot online store, so I assume they are original. I bought them on different occasions, so they are not from same batch. Link: https://store.myairbot.com/flight-controller/omnibus-f7/omninxtf7.html
I also bought spare MPU6000 external gyros and use them with betaflight. 

> 2. Was it working before (in 2.1 or 2.0)?
I never tested boards with old iNav. They do work as expected with betaflight 3.x and 4.x.

Just to re-iterate problem I am having and steps to reproduce gyro issues:

There is no config changes. Board is not soldered, brand new item. Board is positioned with usb to the right.

I flash latest (4.0.4) Betaflight on it and observe correct movement of model in config tool. Switching gyro to second also exhibit correct movement. No orientation changes required for both gyros, everything works out of the box.

I flash latest iNav (2.2.1) and model does not move as expected. Orientation of first gyro is incorrect, requires board or gyro alignment changes of CW270. So this is immediately inconsistent with Betaflight target, and easy to reproduce, there is no config changes whatsoever.  

On top of that, if I install ICM second gyro, it is detected as MPU6500 (unlike BF, where it is detected correctly) and also no accelerometer detected at all.
If second gyro is MPU6000 board does not boot. I now understand this is expected, however it does work fine in Betaflight.

I do appreciate all work you do for community, so do not want to waste your time on this anymore. That second gyro problem render board not usable for me anyway. Will go back to old trusty Omnibus F4 =)

digitalentity, 2019-07-15:
Yep, the board doesn't have the arrow marking "forward" direction, so when the target was created we assumed that USB should be on the left of the craft (as with many other boards). Hence the 180deg off gyro orientation. We can re-align this with Betaflight though.

In INAV when you change the `gyro_to_use` parameter you need to force auto-detection of the accelerometer in the majority of cases (`set acc_hardware = AUTO`).

I'm trying to understand the requirements here so we can fix it, however if we change the default alignment of the gyro it will break stuff for some users. @DzikuVx what do you think? what is the fraction of users who use OMNIBUSF7NXT target?

DzikuVx, 2019-07-15:
In last 3 months, board was connected in 852 sessions and that gives 0,04% of all sessions in that time period @digitalentity

digitalentity, 2019-07-15:
I figure we can fix stuff as impact should be minimal. Thanks @DzikuVx

alex-vasilyev-myob-com, 2019-07-15:
> In INAV when you change the `gyro_to_use` parameter you need to force auto-detection of the accelerometer in the majority of cases (`set acc_hardware = AUTO`).
> 

this is great info, I will try that.

> I'm trying to understand the requirements here so we can fix it, however if we change the default alignment of the gyro it will break stuff for some users. @DzikuVx what do you think? what is the fraction of users who use OMNIBUSF7NXT target?

I do not think changing gyro alignment by itself will help. Problem I had that board was not flyable even if I change alignment via CLI. It was doing instant flip on takeoff. I understand there is some sort of dependency between gyro and acc alignment which was not right? 

Same setup was flying just fine on BF, so this is why I opened this ticket in first place.

Also ability to use MPU6000 as external gyro would be nice. What exactly prevent it from working in iNav and it works in BF?

jpopillion, 2019-07-17:
Guys I am having the exact same board alignment problem and changing the alignment of the gyro and acc in the cli is not having any effect.  My wires are cut to just the right distance and the board is aligned with the usb to the right as all of the build documentation is showing.  Same board works great in beta flight but I am using this craft as a long range gps nav craft and need INAV for that to work effectively. Any help in what I can do to fix this other than re-wire my board with the usb on the left would be helpful.

jpopillion, 2019-07-18:
So I apologize if this was already answered but if I have the board aligned with the usb to the right and the default alignment is with it on the left why can't I just change the board alignment to match how I have it aligned and problem solved.  That is the intent of those variables.  I don't think the problem is necessarily with the default alignment. I think the  problem is I cant use the supplied variables to change the alignment degrees to match my setup and it work.  Again, thanks for all of you guys efforts let me know what I might be doing wrong or is this really a bug?

mikeboon, 2020-03-02:
Has this issue been resolved? I am having the same issue on my OmniNXT F7 on INAV and suddenly it can't boot up and keeps checking for Gyro/Acc giving that 2 long Blue LED flash followed by 3 short flash. I can go into DFU and flash Betaflight 4.0 and it works.

mikeboon, 2020-03-02:
> ## Current Behavior
> There are few issues with latest 2.2.1 firmware for OMNINXT F7 FC.
> 
> 1. Firmware settings for alignment of onboard gyro (gyro_to_use=0) is incorrect. Model does not move as expected. If board alignment changed than model moves ok but quad insta crash because of ACC vs GYRO alignment is wrong.
> 
> Works as expected on Betaflight 4.x. In BF 4 gyro 0 alignment is 0 and gyro 1 alignment is CW270.
> 
> 1. If gyro_to_use set to 1 and second gyro is MPU600 board just hang and cannot boot. Require re-flash.
>    If gyro hardware changed to ICM20602 then board boot but gyro not detected properly (MPU6500 instread of ICM) and accelerometer not detected at all. Changing acc hardware manually does not help.
> 2. RX2 pin is not working. I have CRSF RX connected and it works fine in BF 4.0. After I flash iNav 2.2 receiver is no longer works, but telemetry still does. It seems RX2 pin is not working? Moving RX to UART3 helps, but just curious about pin.
> 
> ## Steps to Reproduce
> 1. Flash iNav 2.2.1 and observe model in setup window. Orientation is wrong.
> 2. Use CLI to set gyro_to_use=1, observe no accel or hang.
> 3. Connect CRSF to UART2, no RX detected while TLM working.
> 
> ## Expected behavior
> Should work as in BF 4.x.
> 
> * FC Board name and vendor: Airbot OMNINXT F7
> * INAV version string:
> 
> INAV/OMNIBUSF7NXT 2.2.1 Jul 3 2019 / 22:26:41 ([a6d8474](https://github.com/iNavFlight/inav/commit/a6d847482a4102973dfbd43030cee8e58893a579))
> GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

Did you manage to solve this?

closed 2022-01-10 by DzikuVx