iNavFlight/inav #4940: External PINIO for MATEKF722mini

romko1, 2019-07-15:
Hello,
I need advice on changing the PINIO in the latest INAV.
I have FC MatekF722mini, which has two PINIOBOX defined:
C08: PINIO1 OUT
C09: PINIO2 OUT
The PINIO is used as follows:
[image]

I want to add an external PINIO to power off the camera.
However, when I connected it to an unused TX pin, and chose eg. RunCam device, so it didn't work like I wanted.
If I connected PINIO physically to pin S6 (B03) and modified it in FW - target.h, it worked (power off the camera) but it also turned off VTX and I don't want it because I use TBS SmartAudio.
[image]

Can you please advise me where to change to VTX not disconnect, ie. it did not disconnect the VBAT power supply for VTX,  because I have the camera connected to + 5V and I only disconnect it via external PINIO.

romko1, 2019-08-29:
Problem solved by firmware modification.

closed 2019-08-29 by romko1

whoim2, 2019-12-09:
> Problem solved by firmware modification.

Hello! Im try compile my target as you for pinio functions

> //PINIO - DAC
#define USE_PINIO
#define USE_PINIOBOX
#define PIN03_PIN PA8
#define PIN04_PIN PA4 // DAC
//END PINIO

but its not work, am use USER1/USER2 modes, CAMERA ON and etc, at pin PA4/PA8 not any changes in levels. You may help?

Inv3nt0r1, 2021-07-09:
> Problem solved by firmware modification.

Can you tell what you did to resolve this?

cryptictuna, 2022-04-20:
can you upload custom hex ?