iNavFlight/inav #4943: Newbie questions

bmalhigh, 2019-07-15:
Hi folks,

Posting some newbie questions. We have a simulation system that has been integrated with PX4 based flight controllers. Looking to add support for iNavFlight now. Hence some questions related to that --

1) Does iNav support HITL and/or SITL?
2) PX4 integration required MAVLink support. Can someone clarify the status of MAVLink support in iNav?
3) Any info on how to hook up iNav with a companion computer like TX2?
4) Has someone done a PX4 to iNav migration involving a Simulation system? Would be great to understand the lessons from that.

DzikuVx, 2019-08-04:
On all questions the answer is no. No this not PX4

closed 2019-08-04 by DzikuVx

dimon777, 2020-01-30:
The best SITL implementation (for beta flight) I could find is: https://github.com/timower/KwadSim
It feeds simulated gyro, accel, etc... data into "virtual quad" running betaflight and getting back motor outputs, simulating movement in 3D space. And although INAV is not PX4 ( and better not to be! :) ) it doesn't mean it shouldn't have HITL or SITL support.