iNavFlight/inav #4945: Matek F405SE Current Sensor Wrong

sanj3k, 2019-07-16:
## Current Behavior
I get inaccurate amp readings around 10A (when through usb powered only) here is a screenshot: [https://i.imgur.com/bVeqVQG.png](url)

With a lipo attached its about 17 amps disarmed and armed 104 (which is the maximum). I have of course checked the cli and the current meter is set to adc so everything should be alright... any other ideas?

## Steps to Reproduce
No need its just everytime wrong

## Expected behavior
Correct amp values

## CLI Dump
https://pastebin.com/TEbMFrb8

- FC Board name and vendor: Matek F405SE Wing FC
- INAV version string: INAV/MATEKF405SE 2.2.1 Jul  3 2019 / 22:48:46 (a6d847482)

DerElch87, 2019-07-16:
I don't understand why a lot of people are opening issues when they only need support.
My first proposal is that you should ask in a community like rcgroups for help.

Opening so much useless issues only keeps the developers away from doing useful things...

To your question.
1. You should flash another version and look at the result
2. I have the same board and firmware running and the current sensor is working fine!

So it is for sure not a bug or issue.
99% a out of tolerance current sensor - can be fixed most likely by setting cli values for the parameters...

sanj3k, 2019-07-16:
I had asked in the rcgroups and even posted a diff but no one responded so it seems like everything is alright. I will flash an earlier version... And what do you mean by setting through cli? Just type in again the same values as cli commands?

closed 2019-07-16 by sanj3k

DerElch87, 2019-07-16:
You should flash an earlier version to check if the values are the same (99% they are the same).
In the configurator or via CLI you can change the parameters for the current calculation:
set current_meter_scale = XXX
set current_meter_offset = XXX