iNavFlight/inav #4947: MATEK F411-WING Softserial RX

CSMachineCo, 2019-07-16:
## Current Behavior
Softserial RX on the LED pad does not appear to work in firmware: MATEKF411_FD_SFTSRL version 2.2.1.  

[image]

## Steps to Reproduce
1.Configure Softserial port for GPS @19200 buad
2.Plug GPS TX in to PAD LED and GPS RX into PAD ST1

## Expected behavior
Receive GPS data as indicated in inav configurator.

## Additional context
Tested GPS connectivity using hardware UART 2 configured with same baud and it works as expected.  Tested with 3DR radio and produced same behavior.  Tested with firmware version 2.2.0 and produced same behavior. GPS device is BN-880.

- FC Board name and vendor:
MATEK F411-WING
- INAV version string:
# version
# INAV/MATEKF411_FD_SFTSRL 2.2.1 Jul  3 2019 / 22:53:44 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

padcom, 2019-09-08:
I am seeing the exact same behavior. I was hoping the software serial port would be good enough to drive the GPS in UBLOX7 mode but in fact it is unusable even in NEMA mode. The systems shows "GPS Failure" every few seconds.

stronnag, 2019-09-08:
We've known since c. 2015 that receiving on soft serial is unreliable. Nothing new here. Don't do it.

CSMachineCo, 2019-09-10:
If it's not a supported function, then it probably should be removed/prevented from/in the configurator...  I don't know how to move issues between projects on GitHub so I'm closing it here. If someone else wants to move it to the configurator, have at it.

stronnag, 2019-09-10:
It is a supported function; just not for your use case.

padcom, 2019-09-10:
OK - can you state what is it supported for? I mean if it is supported for something it'd be great to have it listed so that we don't try to use the software serial port for things that it is not intended to do. I was pretty sure if 19200 is the max baud rate then it doesn't matter what is actually transmitting on this port unless it is conforming to the 19200 8N1 protocol.

stronnag, 2019-09-11:
> OK - can you state what is it supported for? I mean if it is supported for something it'd be great to have it listed so that we don't try to use the software serial port for things that it is not intended to do. 

Extant documentation, https://github.com/iNavFlight/inav/blob/master/docs/Serial.md, specifically:

"UART is the most efficient in terms of CPU usage. SoftSerial is the least efficient and slowest, SoftSerial should only be used for low-bandwidth usages, such as telemetry transmission"