iNavFlight/inav #4948: 2.2.1 Fixed wing permanently enabled launch mode flyaway

tomppak, 2019-07-17:
## Current Behavior
Fixed wing permanently enabled launch mode causes autolaunch to freeze so that aircraft does not respond to any stick or flight mode change inputs. Responds to only disarm switch. When rearmed while in air, issue reoccurs.
[LOG00023.TXT](https://github.com/iNavFlight/inav/files/3400638/LOG00023.TXT)
[LOG00024.TXT](https://github.com/iNavFlight/inav/files/3400639/LOG00024.TXT)

## Expected behavior
As documented.

## Additional context
Not sure if relevant but made an autotune flight and saved settings before issue reoccurred first time.

Dump: https://pastebin.com/hJHAf73U

- FC Board name and vendor: Matek F405 Wing
- INAV version string: # INAV/MATEKF405SE 2.2.1 Jul  3 2019 / 22:48:46 (a6d847482)

expipiplus1, 2019-08-29:
@tomppak, I hope you got your plane back and it isn't still flying! I'm curious to see the DVR footage :)

tomppak, 2019-08-30:
Here`s GoPro footage for you. https://youtu.be/aqPo6mBVtZg

expipiplus1, 2019-09-16:
Thanks, looks as though you could glide it down while disarmed; nice piloting!

Constructively: 

Does it ever make sense to have launch mode enabled while the craft is in the air (and hasn't just been tossed)? Perhaps "always enable launch mode" should be "alway enable launch mode while on the ground"?

Zangetsu57, 2019-12-17:
I just had a similar issue the other day with my HC44, but launch mode isn't permanently enabled.  What happened is the launch failed to detect, but since my idle throttle was sufficient for the plane to fly, it flew away and started to fly in a big circle.  It was unresponsive to controls or any change of flight modes, including RTH.  I waited for the launch timeout of 60s and still didn't regain controls.  Once it was up over a thousand feet in the air (it kept climbing in a circle), I decided to cycle the power on my radio transmitter and see if failsafe mode would kick it out of the state it was in.  Fortunately it worked and I had control afterwards.

breadoven, 2022-11-26:
The log file shows that Launch was never detected so it's assumed this was caused by a high Launch Idle setting allowing a climb after failed detection and no possibility to stick abort with throttle raised. Closing since likely caused by setting error rather than firmware bug.

closed 2022-11-26 by breadoven