iNavFlight/inav #4952: "set osd_video_system = PAL" causing F411 wing from Matek to freeze

Daniel-1276, 2019-07-17:
## Current Behavior
Environment:
Target: MATEKF411
INAV version: reproduced on 2.1.0, 2.2.0 and 2.2.1

Issue:
As soon as I apply power by USB or LiPo to the FC it boots up and blue LED stays constantly on. If this happens, FC is frozen and no serial communication, nor usage for flying is possible. If I repeatedly reboot the FC sometimes (lets say 1 out of 20 tries) blue LED starts flashing again and I can connect and fly without issues.

## Steps to Reproduce
1. Install 2.1.0/2.2.0 or 2.2.1 on MATEKF411 target (other versions not tested)
2. Go to CLI and only set "set osd_video_system = PAL"
3. save and reboot.

## Expected behavior
PAL mode for OSD should be set and of course FC should boot up normal.

## Suggested solution(s)
If I have set PAL mode I do have to power cycle the FC a dozen times and sometime it boots up with blinking blue LED and all is fine. If I do have the booting issue and manage to get INAV configurator to connect, I can set back osd_video_system to AUTO and FC behaves completly normal. Somthing seems wrong with PAL mode at least on my target.

## Additional context
- FC Board name and vendor: Matek F411 Wing 
- INAV version string: # INAV/MATEKF411 2.2.0 Jun 18 2019 / 14:54:48 (5e5551bd3)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

Jetrell, 2019-07-17:
I found the same thing. When I used either of the softserial firmware's. 
It would happen when the USB was connected. 
I went back through my DIFF ALL files and systematically narrowed it down to the  `set osd_video_system`. Which I also had it set to PAL.

Daniel-1276, 2019-07-19:
@Jetrell Interesting, so this is clearly reproducible. Wonder if this happens only on Matek F411 wing FC ? It uses the same AT7456 OSD chipset than all other modern FC's. At the moment I dont't have an issue, setting it to AUTO, but took me a while to figure out the root cause.

Jetrell, 2019-07-19:
That was the same here.  At first I thought it was a configurator issue with the F411-Wing FD_SFTSRL firmware.  
It was only later, that I track it down to - `set osd_video_system = PAL` being used on the OSD.
I had the same settings loaded with 2.1.0, and it worked fine. But I was only running normal F411 firmware with it.

I haven't gone back to look any deeper into it. But 2.2.1 with all F411 firmware's seems to be fine, IF `set osd_video_system = AUTO` ... Like you have found.
So it could be - SFTSRL2 and FD_SFTSRL firmware's causing it. 
Not sure what that has to do with the video?

@shellixyz Could you look into this for us?

AnandNL, 2019-07-22:
the f411-RSSI firmware shares this problem and the auto fixes it for now.

DrBonesLCD, 2019-09-10:
My problem has been that as of 2.2.1, my camera would shut down leaving me with the OSD only, or the camera would work and no OSD. Connecting it to the battery multiple times would eventually lead to both working...but I lost the camera twice (i.e. gray screen and only OSD) twice during flight. My setting was: osd_video_system = NTSC. I set it to auto, and it seems fine now. This behavior has been observed by someone else on rcgroups here: https://www.rcgroups.com/forums/showthread.php?3153930-OSD-overlay-OR-Camera-not-both-at-same-time%21#post40423691

closed 2023-03-06 by DzikuVx