iNavFlight/inav #4953: Virus warning from windows defender

EdSmega, 2019-07-17:
I am a new INAV user.  I installed it on 2 windows 10 computers and it worked on both until the latest Microsoft updates were applied to my desktop machine.  Inav stopped working and I got error messages about a virus.  Thinking I may have downloaded accidentally from a bad site, I deleted it and attempted to reload from github.  Now Windows defender won't allow the download to complete.  Message says threats found.   Before I deleted the previous install, the messages were to the effect that The software was downloaded from one of those bogus downloaders that bundles crap with your files.  Sorry I don't have the exact text anymore. attached is the screenshot from where I downloaded.    Error message from the notification error disappears when I activate my screen capture.

[image]

stronnag, 2019-07-17:
You should raise configurator issues in the configurator issue tracker.

You should (probably) raise Windows issues with Microsoft; we don't have the source code to fix them.

EdSmega, 2019-07-17:
Sorry for any breach of protocol.  I am a brand new Github user and haven't learned my way around yet.    I don't really understand what you suggest I should do.  Is there an easy way to move my original post to where you suggest it belongs?