iNavFlight/inav #4956: Inav 2.2.1 on PWM6 for Omnibus F4 V2.1 Flight Controller Clone

c172jeff, 2019-07-18:
Hi, 
   I have a clone Omnibus F4 V2.1 Flight Controller STM32 F405 MCU, Integrated OSD that is missing the CH5 and CH6 pads.  They appear to be missing from the location where they are supposed to be.  The board appears the same otherwise.  I want to use smart audio and software serial.  It looks like I cannot?  Is it possible to do  some kind of resource mapping of PWM6?  If not, can someone provide me a version is mapped to PWM6 for software serial tx so I can use smartaudio.  

Thanks,
Jeff

teckel12, 2019-07-18:
@c172jeff Could you provide a picture of the board?  There's not resource mapping in INAV so it would need to be a custom build.

c172jeff, 2019-07-18:
Thanks.

The board is in my Nano Talon.  This is the second board from Banggood.  The first is in a tree.

On the top of the board, the CH5 and CH6 pads should be near MOTO (near red wire).  They are not.

On the bottom of the board, the RSSI pad was damaged by me, and is missing.  This is in the same location of the first board/where it should be.  Also, I tested the three pads below the SD card.  These seem to relate to the USB connection.

[image]

[image]

Jeff

c172jeff, 2019-07-20:
Does anyone have a build for this?  I'd like to get to get this plane together and use smart audio with this board on PWM6?  TBS Unify Pro

c172jeff, 2019-07-21:
I have downloaded the source code and can recompile this version on my computer.  If no one can make a custom build, what files might need editing to get it on PWM6?  I heard timers might make it not possible.  Also, how would one unmap what the current PWM6 is linked to?  Thanks.Jeff

DzikuVx, 2019-07-24:
@c172jeff look for `target.c` and `target.h` files in `src/main/target/OMNIBUSF4` folder.

closed 2019-07-24 by DzikuVx