iNavFlight/inav #4957: allow a lower climb angle for navigation modes

Mckenzie74, 2019-07-19:
## Current Behavior
with nav_fw_climb_angle set to its min allowed value of 5 my plane still climbs too much during waypoint missions for my needs: 120 meters per km
## Desired Behavior
To increase efficiency on long range missions I'd like to set it to 4 or 3 or even less, but at the moment the min allowed value is 5. I'm perfectly aware that rth will be affected as well but I'm ok with that and the defaults should stay the same as they are today. Just let me go lower if I know what I'm doing.

## Who does this impact? Who is this for?
long range efficiency lovers
Thank you!

wx4cb, 2019-07-19:
I wonder if you couldn't  lower the cruise throttle or the speed to achieve that? it might be also that you have too much of a height difference and will need that angle. 

if not then also reducing the height difference would probably lower that too

Just some thoughts

giacomo892, 2019-07-20:
Have you tried playing with nav_auto_climb_rate ?

Mckenzie74, 2019-07-20:
@wx4cb thanks for your thoughts. Cruise throttle is already at such a level that would be risky going lower. During missions inav will always reach next waypoint at nav_fw_climb_angle if it's higher or nav_fw_dive_angle if lower, no matter what the height difference is. I wish it could climb and dive like it does in the new linear_descent_rth, but as of today it cannot.

@giacomo892 nav_auto_climb_rate is working only for multirotors, mine is a fixed wing.
What I could try is lowering nav_fw_pitch2thr from the default 10, but in the past I had a bad experience when playing with that value, now I'm a little scared to touch it and always leave it at default.

giacomo892, 2019-07-20:
@Mckenzie74 you are wrong here. It works on FW too. But unsure if it works on WP mode too, don't have time to check the code. 
And no, don't touch nav_fw_pitch2thr.

Mckenzie74, 2019-07-20:
you sure? I guess it will not hurt trying anyway. I'll let you know about nav_auto_climb_rate

Mckenzie74, 2019-07-21:
@giacomo892 Today I tested nav_auto_climb_rate=100 (half of what it was the flight before), theorically a desired vertical speed of 1mt/sec, but nothing changed both in waypoint and 3d cruise. Z84 climbed again at 120 mt per km and 2 mt/sec.
I definetely think that nav_auto_climb_rate applies only to MRs and the only way to smooth out wp climbs is by lowering nav_fw_climb_angle

giacomo892, 2019-07-21:
That setting for sure doesn't work in CRUISE3D since YOU are commanding the ascend and the descend. For that you need to tune nav_manual_climb_rate

Mckenzie74, 2019-07-21:
So auto_climb is not for missions nor 3dcruise. I can’t see where it could be useful since every other mode has throttle manual control.
My problem is for wp missions and my test flights lead me to narrow it down to nav_fw_climb_angle only. If devs could allow a lower number I’d be happy, but I will fly inav happily anyway

hali9, 2020-02-24:
I think it can be closed by https://github.com/iNavFlight/inav/issues/5219

closed 2020-02-24 by giacomo892

breadoven, 2020-10-01:
From the code  nav_auto_climb_rate is only used for a FW in conjunction with nav_rth_home_altitude.  So just controls the rate of climb/decent when adjusting to that RTH home altitude.