iNavFlight/inav #4959: CL Racing F4 Air V3 Target Missing

umdpru, 2019-07-20:
I see the CL Racing F4 Air V3 Target in the master but it's not being compiled? How come?

bnn1044, 2019-08-08:
I am not sure why the target file is not included in the release. the code is in the master. 
@digitalentity  may know?

thanks.

keithluneau, 2019-08-08:
I'd like to see this sorted out too, I have this board here waiting for firmware right now. I would attempt to compile it myself but that's a bit outside my comfort zone if I'm being totally honest... lol I wouldn't know where to start.

teckel12, 2019-08-09:
@keithluneau @umdpru Since I was already doing some compiling anyway, here's the firmware for the CLRACINGF4AIRV3:

[inav_2.2.1_CLRACINGF4AIRV3.zip](https://github.com/iNavFlight/inav/files/3487049/inav_2.2.1_CLRACINGF4AIRV3.zip)

keithluneau, 2019-08-09:
Thanks! I also found it here for download - https://cl-racing.myshopify.com/pages/support

closed 2019-10-11 by DzikuVx

mueller2030, 2020-03-07:
Hi, 
Is there an updated version of this target? i have a new  CLRACINGF4V3 and the target is not provided in the drop down menu as well as on the clracing website, they are both out dated and wont run on the latest INAV config
thanks