iNavFlight/inav #4963: Artificial Horizon wrong movement?!

methisto, 2019-07-22:
Hi Guys!

Before i do my first flight with the SkySurfer i have a Question:

When i move the Plane to the left side, the Horizon Bar in the OSD goes to the right, and when i move to the right, the Bar goes left?!
In the Setup Tab when i move the plane everything seems correct, the F405 Wing is mounted correct direction, and the High Five Test and the Way the Servos act when i move the plane seems to be correct?!

And another issue: I have a XM+ receiver, and its really "swamping" When i move the Radio (QX7) more far away, but when im just about 1m away it loses connection, on my Tyro99 also with XM+ i dont have this issue!

Any suggestions?
Thanks!

giacomo892, 2019-07-22:
Not the right place to ask such questions. 
You can use board allignment to level out the board after the calibration.

closed 2019-07-22 by giacomo892

digitalentity, 2019-07-23:
@methisto you would get responses from a bigger community here https://t.me/INAVFlight