iNavFlight/inav #4965: Omnibus F4 v3 GPS toilet bowling

jasonchen31, 2019-07-22:
## Current Behavior

First of all, thanks a lot for all of the developer's hardware. Here I would like to report a POSHOLD bug. My setup is as below:
FC: Omnibus F4 v3
INav: 2.2.0
GPS+Compass: Beitian 880.
I suffered a lot from the toilet bowling issue for 1 week. I searched all forums and threads. All people said it the toilet bowling issue results from bad compass calibration. I did tried dozen of times all kind of calibration methods and lifted the compass almost 20cm away from everything. I also tried removed prop, spin the motor and checked the compass reading. The compass was almost perfectly calibrated without any impact from the motor or power lines. But all tests resulted in toilet bowling. The only difference is that the better calibration the slower toilet bowling started.
After days of calibrating and testing, I gave up trying and downgrade the INav version 2.1.0. And suddenly POSHOLD works perfectly !!!!
Therefore, I concluded that the POSHOLD feature in v2.2.0 is buggy. Maybe it was caused by the newly introduced FF PID feature. 
FYI.

## Expected behavior
Position hold

## Suggested solution(s)
downgrade to v2.1.0 

---

v2.1.0
- FC Board name and vendor: OmnibusF4v3, HAKRCF4 mini, 
- INAV version string: 2.2.0
- GPS+Compass: Beitian BN-880

teckel12, 2019-07-22:
@jasonchen31 Do you have your compass orientation set to 270 Flip and the GPS antenna pointing towards the sky?

DzikuVx, 2019-07-23:
@jasonchen31 despite what you belive, toilet bowling is magnetometer connected is caused by bad orientation and/or calibration. When BN-880 is facing upwards with cables backwards then it is 270FLIP. Set it, calibrate, check if heading reported matches the reality.
It is also possible that magnetometer is damaged and is just reporting bad values.
Either way, this is not a bug but hardware issue and I can confirm podhold workin on BN880 in 2.2 since I'm using this and it works perfectly

https://www.youtube.com/watch?v=FlEm0-pXNf0

closed 2019-07-23 by DzikuVx

jasonchen31, 2019-07-23:
Sorry for late response. I did have CW270FLIP setting. And I did all the calibration with that setting. I'll try again with default GPS pre-setting again with 2.2. If it still exist, I'll reply here. Thanks for your comments.

teckel12, 2019-07-23:
@jasonchen31 Be sure to do calibration away from large objects like cars or in you house. Heck, I've had calibration issues due to concrete.

I also can confirm that POSHOLD works in 2.2 without issue.

jasonchen31, 2019-07-23:
Dear Teckel12,

Thanks a lot for your support. I migrated to 2.2.1 again by using diff from 2.1.0. And, WOW, NO more toilet bowling!!! I doubted that the failure of my first iNav installation with v2.2.0 was caused by the default 'gps drone' pre-setting? In order to contribute a little to the community, I flashed v2.2.1 and configured everything from the very beginning based on the presetting. But unfortunately I cannot repeat the toilet bowling issue anymore. Now everything works fine!!
Many thanks for all of your contribution! you did a quite good job! I appreciate your prompt response very much!

Success!!
Jason