iNavFlight/inav #4966: Bad behavior in ALTHOLD mode, 2.2.1, MATEK F405-CTR

xrife, 2019-07-23:
Hi!

I'm a newbie. Faced a problem in ALTHOLD mode after set throttle stick turns to the lowest position.

## Current Behavior
It is strange that, with the gas turned off and no acceleration, the baro height continues to increase, although this was not so visually.

After raising the stick of gas to small values, the motors begin to roar and the quad rushes up.

## Expected behavior
After returning the throttle stick from the lower position, the quad should change the target altitude value or go towards it slowly, as specified in the settings.

## Additional context
[pastebin](https://pastebin.com/ajbC67Ni)

- FC Board name and vendor: MATEK F405-CTR
- INAV version string: 2.2.1

xrife, 2019-07-23:
[LOG00054.TXT](https://github.com/iNavFlight/inav/files/3422465/LOG00054.TXT)
[image]

xrife, 2019-07-26:
It looks like #4925 
What do you think? What could be the solution in my case?

teckel12, 2019-07-27:
@xrife I believe this is typically caused by vibrations.  Is your FC hard or soft mounted?

xrife, 2019-07-28:
> @xrife I believe this is typically caused by vibrations. Is your FC hard or soft mounted?

Soft mounted as recommended.

Modes ANGEL and HORIZON and ACRO (without ALTHOLD) work fine even at maximum throttle. The current problem occurs when the gas is turned on after it is turned off in flight with ALTHOLD&ANGEL.

teckel12, 2019-07-28:
@xrife Keep in mind that alt hold and pos hold works fine for me no matter what I do with the throttle.

xrife, 2019-07-29:
Thank you for your experience and comments.

My settings now by `dump diff` [https://pastebin.com/6aEwb3xD](https://pastebin.com/6aEwb3xD)

I'll try set `nav_use_midthr_for_althold` to OFF and `set nav_mc_hover_thr` to my correct mid throttle position and share my experience.

xrife, 2019-08-04:
Hi!

I conducted an experiment with these settings, but got the same result.
Maybe my problem is in my settings? If someone shares their settings, I can check the work.

[image]
[LOG00094.TXT](https://github.com/iNavFlight/inav/files/3464949/LOG00094.TXT)

I can assume that the problem is in my settings, since accelerations coincide with visual observations. But the speed and position do not coincide with the acceleration integral and visual observation.

xrife, 2019-08-17:
@teckel12 The problem only appears when AIRMODE is disabled. If AIRMODE is enabled, then ALTHOLD works fine.

teckel12, 2019-08-17:
@xrife I always have airmode enabled (enabled via the arm switch).

I can try to set it via a different switch and see if I can duplicate your issue.

xrife, 2019-08-19:
@teckel12 I switched AIRMODE to a debug switch to turn off the motors in flight without DISARM.

I will be glad if you try this for your setup in order to understand where to go next.

gilglecio, 2019-08-29:
I have the same problem, I also use this controller, I am a week looking for solution, my drone is varying in ALT HOLD one meter up and one meter down

closed 2022-02-23 by DzikuVx