iNavFlight/inav #4968: led strip on omnibus F4 v3 not working

raffxj, 2019-07-23:
hi

Ive flashed inav 2.2.1 softserial m5m6. confirmed working for rangefinder connected to it.

ive used the gui to enable ledstrip and also 'feature LED_STRIP' on CLI.. after reboot, still no led on ledpad. 

how do I 'activate' this?

af6767, 2019-08-09:
Hi,

i have the same fail .
After looking in source code, for me i'm missing an additional fix in the target.h. Otherwise the led on V3 Boards are dead.

at line 261 you find :
#define USE_LED_STRIP
#if (defined(OMNIBUSF4PRO) || defined(OMNIBUSF4V3) xxxxxxxxxx) && !defined(OMNIBUSF4PRO_LEDSTRIPM5)
#   define WS2811_PIN                   PB6
#else
#   define WS2811_PIN                   PA1
#endif

put this at xxxxxxxxxx :
|| defined(OMNIBUSF4V3_S6_SS) || defined(OMNIBUSF4V3_S5S6_SS) || defined(OMNIBUSF4V3_S5_S6_2SS))

i have currently no time to test it, but for me it will fix it.

cu
Andreas

raffxj, 2019-08-10:
are there any addtl info how I would go about compiling them for windows 10?

datamann, 2019-08-10:
@raffxj Not sure if this will help you but you can take look at my comment on how to set up Docker for compiling INAV code at this thread #3896

af6767, 2019-08-13:
another way and in my opinion the easiest way, in Windows 10 you can install a linux subsystem. Google for Linux subsystem in Windows 10 and get a install instruction. I would suggest Ubuntu.

datamann, 2019-08-13:
I just tried what @af6767 said and I have to agree. When I found this https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md it got a lot easier. You can still use windows code editor e.g. VS Code sinse the INAV code files are available from windows filesystem.

LawrenceLDJr, 2019-09-14:
Use Omnibus F4 V3 SS M6, I use M6 for Smart Audio but but typed resource into CLI and saw LED on M5 pad and it works.

breadoven, 2022-11-26:
Seems this should have been closed.

closed 2022-11-26 by breadoven