iNavFlight/inav #4972: Unable to get any GPS mode to activate

nick9one1, 2019-07-24:
I can't seem to get any GPS mode to activate. 

I have all green ticks, fully calibrated, at least 10 sats. I can arm and fly, but when I switch into RTH, POSHOLD, or NAV WP nothing happens?

`# version
# INAV/MATEKF405 2.2.1 Jul  3 2019 / 22:31:06 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# logic
logic 0 0 0 0 0 0 0 0
logic 1 0 0 0 0 0 0 0
logic 2 0 0 0 0 0 0 0
logic 3 0 0 0 0 0 0 0
logic 4 0 0 0 0 0 0 0
logic 5 0 0 0 0 0 0 0
logic 6 0 0 0 0 0 0 0
logic 7 0 0 0 0 0 0 0
logic 8 0 0 0 0 0 0 0
logic 9 0 0 0 0 0 0 0
logic 10 0 0 0 0 0 0 0
logic 11 0 0 0 0 0 0 0
logic 12 0 0 0 0 0 0 0
logic 13 0 0 0 0 0 0 0
logic 14 0 0 0 0 0 0 0
logic 15 0 0 0 0 0 0 0

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
[... 1089 lines cut ...]
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 45
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 3
set vbat_cell_detect_voltage = 425
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH
`
[image]
[image]

DzikuVx, 2019-07-24:
NAV modes activate only after arming, never on the ground before arming.

nick9one1, 2019-07-24:
> NAV modes activate only after arming, never on the ground before arming.

Yep, I said this in my first post. 

> When in the air if I switch into RTH, POSHOLD, or NAV WP nothing happens?

DzikuVx, 2019-07-24:
Please provide blackbox log and DVR receording if possible

nick9one1, 2019-07-24:
[LOG00002.TXT](https://github.com/iNavFlight/inav/files/3427112/LOG00002.TXT)
Sure, Here you go. I'll try and get a video too.

stronnag, 2019-07-24:
* The craft reports entering PH and WP modes
* There are 0 waypoints
* There is no baro, so it's moot anyway.

nick9one1, 2019-07-24:
It was the baro - I forgot to enable it in the config. Everything is working now. 

If none of the GPS modes will activate without a baro, maybe this should report an error? 'Unable to activate mode' 

e.g I'm sure if I didn't have GPS configured it wouldn't let me activate the mode? Should be the same for baro?

closed 2019-07-25 by DzikuVx