iNavFlight/inav #4976: OMNIBUSF4 fails to detect I2C BMP280 since iNav 2.2.0

hogthrob, 2019-07-26:
## Current Behavior
BMP280 via I2C not detected on 2.2.0 on an OMNIBUSF4 target.

## Steps to Reproduce
1. Connect BMP280 correctly via I2C
2. Flash iNav 2.1.0 and make sure BMP280 is detected
3. Flash iNav 2.2.0 (or 2.2.1)
4. BMP280 is not detected anymore

## Expected behavior
BMP280 detected

## Suggested solution(s)
The commit https://github.com/iNavFlight/inav/commit/a973acc1a0cb2d3755736db944830cb35596bad8
needs to be rethought. It basically disable I2C as the default way of accessing the BMP280.

The "combination" of SPI and I2C as defined in
https://github.com/iNavFlight/inav/blob/956a3f5ed9aa634a7930a5e4229119f0773a2dde/src/main/target/OMNIBUSF4/target.h#L118
does not work, only SPI works if BMP280_SPI_BUS is defined.

If I comment out the BMP280_SPI_BUS, the BMP280 is detected again. 

## Additional context

---

- FC Board name and vendor: GOTOQOMO Betaflight Omnibus F4
- INAV version string:

rozhkovets, 2019-07-26:
I confirm that BMP280 is not detected on 2.2.1 on an MATEK F405-CTR.

shellixyz, 2019-07-26:
Works fine for me with 2.2.1 and Matek F405-CTR but I confirm the issue was present in 2.2.0. It has since been fixed. If you're running 2.2.0 upgrade to 2.2.1

shellixyz, 2019-07-26:
@hogthrob Please try this build it should fix the issue: [inav_2.2.1_OMNIBUSF4.hex.zip](https://github.com/iNavFlight/inav/files/3435963/inav_2.2.1_OMNIBUSF4.hex.zip)

hogthrob, 2019-07-27:
Hi,
thanks a lot for the fix. I will try the build. I am confident it will do the job. I made my own test build before reporting which solved the issue in a similar way, and it did work.

hogthrob, 2019-07-27:
@shellixyz : Yes, this works. However, I now noticed, my SPI flash is gone, too. I had to add another two references to OMNIBUSF4 to get it to work as before (see my referenced fix above). However, I am afraid, these extra changes break the SD Card support for other machines with OMNIBUSF4 target. It seems that at least the MATEK F404-CTR has an SD card connector. 

Not sure how we get out of this.

shellixyz, 2019-07-29:
I'll take a look. The MatekF405-CTR is unrelated to the OMBF4. The OMBF4PRO has a SD card slot

shellixyz, 2019-07-29:
@hogthrob This should fix the SPI flash too, please test: [inav_2.2.1_OMNIBUSF4.hex.zip](https://github.com/iNavFlight/inav/files/3443196/inav_2.2.1_OMNIBUSF4.hex.zip)

hogthrob, 2019-08-03:
@shellixyz: Yes, the last build from you recognizes my SPI flash too. 

I now have identified more changes for the OMNIBUSF4 target build from 2.1.0 : 

The ACCEL / GYRO orientation changed (CW180 to CW270). Luckily I noticed that before the first 2.2.x flight (I was on default for the ACCEL/GYRO orientation).

I don't know if this was intentional. It is easy enough to fix but it may come as a surprise if someone upgradesr inav following the instructions and goes straight to flight...

PR #4977 (2019-08-13): Fix baro and SPI flash on OMBF4
Closes #4976
closed 2019-10-01 by DzikuVx