iNavFlight/inav #4978: cannot connect omnibus f4 in inav, tried every fix available

Noel2351, 2019-07-26:
i have a omnibus f4 board in my flying wing, i was able to originally connect it to inav and flashed it , but now i cant log back into my settings in inav i have tried the zadig driver update, the impulce rc driver fixer puts it in boot mode but inav does not want to flash to board, when i unplug board and plug back after driver fixer inav detects board and version installed even the com port, but fails to connect, i have tried the driver update fixes pretty much read every post and watched every video but nothing helps, even tried flashing betaflight to see if that will reset board but nothing will connect to board, in bootloader mode or in normal mode, im running win 10 pro 64bit.

umdpru, 2019-07-26:
what version of inav do you have? I believe after v 1.9.0 then you need to use the new configurator, if you're on 1.,9.0 or previous then download an older configurator exe from the github and use that to flash. Then use the newer configurator. 

https://github.com/iNavFlight/inav-configurator/releases

Noel2351, 2019-07-26:
i have version 2.2.0 chrome configurator, dont have a offline version

Ralf-Os, 2019-07-26:
Chrome configurator isnt developed any further, just Download configurator
from github

FPVZaphod, 2019-09-04:
I had the same issue with an Omnibus F4 Pro. 
Disconnect ALL Uarts, then Power it up with Boot button pressed, then flash it "without reboot" Option. This worked fine for me.

Noel2351, 2019-09-04:
I managed to flash my omnibus f4 eventually using a old laptop, my desktop keeps kicking it out, also i found that the newer firmware was buggy, my  ppm did not pick up, eventually used i bus and that worked.

closed 2019-09-06 by DzikuVx