iNavFlight/inav #4979: Update Navigation modes>>ALTHOLD documentation

MikeNH, 2019-07-26:
## Current Behavior
I'm not sure where this issue should be to get addressed. I would update the ALTHOLD wiki 
https://github.com/iNavFlight/inav/wiki/Navigation-modes#althold---altitude-hold
if I knew where to get the answers. I'm happy to do what I can. 

I have a new 250 quad build. I'm trying to troubleshoot the problems with Altitude Hold. The Navigation modes>>ALTHOLD wiki has troubleshooting process that should be the first step. Here are some of the problems I ran into. Again, I will correct where I can.

**"In general you shouldn't mix up ALTHOLD and ACRO/HORIZON..."**
Does this mean that ALTHOLD is not compatible with acrobatic maneuvers? While in ALTHOLD aerobatics can't be performed.
 
**"set nav_max_climb_rate = 500"**
When executing the cli "get" command the result is "Invalid name". This variable can't be found in Cli.md.

**ALT P, ALT I, VEL P, VEL I, VEL D**
Although the doc describes these variables, where can they be seen or set? Cli.md? Cli "get" command? 

**1.insufficient ALT_P and/or ALT_I**
"Invalid name" when executing a cli "get" command. Also, not in Cli.md.

**"Let's make a small experiment...set iNAV_w_z_baro_p = 0.5..."**
Cli is case sensitive resulting in "Invaild name" for  iNAV_w_z_baro_p. It is correct in Cli.md.

**"set nav_alt_p = 0" and try flying.**
This variable doesn't exist in cli or Cli.md

**As a second step you can try zeroing out "nav_vel_p" and "nav_vel_i" and "set nav_vel_d = 100".**
nav_vel_* doesn't exist.

**"To deal with oscillations you can try lowering your "nav_alt_p", "nav_vel_p", "nav_max_climb_rate", and "nav_manual_climb_rate"."**
Only nav_manual_climb_rate exists.

_This appears to have great potential as a document for someone (me) trying to figure out why ALTHOLD is not working. I haven't looked beyond this part of the doc because at the moment, ALTHOLD is the challenge_

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

# version
# INAV/MATEKF722SE 2.2.1 Jul  3 2019 / 22:44:40 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

- FC Board name and vendor:
- INAV version string:

MikeNH, 2019-07-27:
I'm responding to the "Feature request" label assigned by the bot. This is a request to correct existing documentation rather than a feature request.

stronnag, 2019-07-27:
Issues are for software bugs, firmware documentation and feature requests. The wiki is a community resource, and as you've seen tends to bit rot. It can however be updated by anyone without going through the review process required by the firmware and its documentation The wiki is not maintained by the development team; it is howver a great place for non-developers to contribute.

You might try asking your questions in the Telegram support chat or the RC Groups forum.

MikeNH, 2019-07-27:
Thank you. That's the suggestion i was looking for.

closed 2019-07-27 by stronnag