iNavFlight/inav #4980: disarm on signal loss (Sbus Orange R620X)

jfmrod, 2019-07-27:
I've been setting up my quad with iNAV to RTH on signal loss.
Position hold and RTH modes work well.
The problem I've been having is on switching on signal loss. As I've reported in the title, I'm using an orange R620X receiver wired through an sbus connection to the FC (Matek F722 STD).

From what I've read it seems this RX does not set a failsafe bit on the sbus connection (is there anyway I can verify this using inav logs?) which prevents iNav from knowing there was signal loss and entering failsafe mode.

What happened in my first try turning off the transmitter to simulate signal loss, was that the disarm switch must have gone low (RX failsafe switch position) and that disarmed the quad in midair (I have disam_kill_switch = ON).

In trying to figure out the problem I saw some suggestions to configure the RX for a hold failsafe but unfortunately I didn't manage to get that working, and only have managed to get it to do a preset position failsafe. Naturally I don't think it is safe to configure an armed and high throttle as the the RX failsafe preset. 

I saw another suggestion to use a throttle very low (lower than normal range) as a signal for inav to enter failsafe mode. I confirmed on the bench that iNav interpreted this as a failsafe signal (by the red failsafe parachute when I turned off the transmitter, or just by moving the throttle stick lower than the normal range). However, when I tried it in the air, I still got a midair disarm.

I suspect there is still some kind of timing issue. Is it possible that the throttle going out of range does not immediately switch the FC into failsafe, and therefore inav still disarms due to the disarm switch?

Appreciate any help on this, is there any simple parameter you think I could change that might fix this issue? I didn't have a spare sdcard for a log but I'll look for one and try making a log.

teckel12, 2019-07-27:
@jfmrod Can you set the RX to send no signals?  Typically, a RX can be set to send no signals, set all channels to low, or allow for a failsafe mode where each channel can have a set value.  When using INAV, it's best to have the RX send no signals.  That way, INAV can take over and do it's own failsafe.

teckel12, 2019-07-27:
@jfmrod Just looked at the manual for the OrangeRx R620.  It doesn't have the option for no signals on failsafe.  This is common with old Rx before we had flight controllers.  I would get a new Rx and you'll probably have issues using this Rx with a flight controller.

closed 2019-08-04 by DzikuVx