iNavFlight/inav #4984: Offset in FC position

holoduke, 2019-07-30:
I try to find and answer related to proper aligning a flight controller.
For a bigger fixed wing (200cm) I would like to position my matek 405 wing a bit to the rear. About 25-30cm backwards from CG.
Would this lead to issues? GPS is located near CG. I see that there is no option to set a longitudinal/latitudinal offset for the FC.

teckel12, 2019-07-30:
@holoduke Why would it be wrong to have the flight controller in the rear?  Are you thinking an adjustment in GPS latitude/longitude would be required for 25cm when a GPS is only accurate to maybe 100cm at best?  Maybe I'm confused by why you believe it would be a problem to have your GPS and flight controller not on top of each other.  I've never had my GPS in the same location as my flight controller.

holoduke, 2019-07-31:
I mean when FC is not in cg, the lateral and vertical movements are more severe and might confuse the FC. If I am right

teckel12, 2019-07-31:
@holoduke I would think closer to COG the better, but I believe it doesn't make much difference if it's a small amount off.  The angles would all be the same no matter where you position the FC.  The only difference would be a slight altitude change if not at the COG.  But I can't imagine an inch here or there would make a difference or even be detectable.  FW pilots would need to confirm this however.

teckel12, 2019-07-31:
@holoduke This issue system is for INAV issues. For assistance you should use the support channels (like Telegram). See the main readme page for support links.

DzikuVx, 2019-08-01:
There is no need to offset anything. Perfect position for FC is CoG but both gyro and accelerometer will work if not in CoG. For GPS it does not matter at all with the scale RC works with

closed 2019-08-01 by DzikuVx