iNavFlight/inav #4986: SPRACING3 does not working with PWM reveiver

PZotya, 2019-08-03:
## Current Behavior
Upgraded from Inav 2.1 to 2.2.1 and the prviously working PWM receiver is not working, the Configuratos shows No Receiver. Tried to set in CLI, the result is the same.

## Steps to Reproduce

1. Upgrade firmware to 2.2.1
2. Set the receiver type to PWM
3. Reboot
4. The Receiver type is NONE

- INAV version string:
---
# version
# INAV/SPRACINGF3 2.2.1 Jul  3 2019 / 21:50:13 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]- FC Board name and vendor:

stronnag, 2019-08-03:
iNav no longer supports PWM receivers. This is a feature. https://github.com/iNavFlight/inav/pull/4949 updated the documentation.

closed 2019-08-03 by stronnag

Explorer200, 2019-09-13:
сбас и остальные протоколы тоже? только PPM поддерживает? что за дичь... выбрать можно только PPM.

digitalentity, 2019-09-13:
Upgrade Configurator to latest 2.2.1 and try again.
https://github.com/iNavFlight/inav-configurator/releases