iNavFlight/inav #4987: New target request: Mamba F722S

drewzh, 2019-08-04:
Product page:
https://www.diatone.us/collections/electronic-system/products/mamba-f722s-stack

Reviews:
https://oscarliang.com/diatone-mamba-f7-stack/
https://blog.dronetrest.com/all-new-f4-and-f7-mamba-flight-stacks/

This board doesn't seem to be supported currently and I'm not aware of any ongoing work to add support?

The board is quite compelling due to the addition of integrated bluetooth.

Many thanks

digitalentity, 2019-08-04:
@drewzh please see https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

drewzh, 2019-08-05:
@digitalentity My apologies - in that case, I doubt I can help push this along as I have limited time available. I'll close this request.

closed 2019-08-05 by drewzh