iNavFlight/inav #4988: Matek F722-STD - Raw Gyro Sampling Sensibility

Elbarto, 2019-08-04:
## Current Behavior
After setting the debug_mode to GYRO with CLI commands, I looked at the raw gyro log data with the blackbox viewers to setup my filter. And there I saw a weird behavior with the pitch axis of the gyro. It's looking like the sensitivity for this axis is really bad (I'v the feeling to read an 8 bit sensor or even less). I took a screen capture to show you the problem :

[image]

I've some vibration on the roll, a bit more on the Yaw but I know where it come from. But the Pitch (blue one) show weird staircase pattern.

That doesn't stop me from flying the drone pretty well because the filter are smoothing all of that, but it definitively introduce error and make the PID tuning more difficult.

I don't know if it is related but I can't find a way to have a looptime greater than 1Khz with this FC. Even when I toggle off the gyro syncing. Is this limitation on purpose ?

## Expected behavior
A smooth curve  for Pitch gyro data (at least not in staircase).

## Suggested solution(s)
I'm not sure where the problem come from. Is the logging rate too high and so I'm missing data frame ? I don't think so otherwise I would have problem on other axis too.
The sampling frequency of the gyro ?
Hardware issue with the ADC of the IMU ?

## Additional context
Here is the dump of my setup :
https://pastebin.com/RHBDnrAG

---

- FC Board name and vendor: Matek F722-STD
- INAV version string: 
INAV/MATEKF722 2.2.1 Jul  3 2019 / 22:40:11 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

In conclusion I think I'll go back to the Matek F4 boards series that seems to be better supported and have better performance. Never had such problem with those.

Thank you for any help, ask for more info.

digitalentity, 2019-08-04:
Is this an issue about visual representation of the data or about flight performance?

Elbarto, 2019-08-04:
Well, I don't "feel the problem" while flying but as I said, after filtering the curve is smoothed so it shouldn't alter that much the flight performance. It may be only a visual bug, I can upload the log file if any one want to try.
The thing is, if it's not a visual bug, even if it doesn't alter flight performance in a human perceivable way, it introduce error and vibration through the system. So it can damage motor or esc in the long term, and add mechanical vibration in the frame. (I'm using this drone for "professional" shooting, so i'm trying to have the fewer vibration possible). After some closer look, it seems that the Pitch gyro data is "updated" at a rate of 62Hz, and I see a spike around this frequency in the spectrum analyzer.

And actually I think it has quite a moderate impact, because I'm having trouble to adjust PID for the Pitch axis, whereas the Roll was easy to set right.

I'll upload log and screenshot in my next post.

pbartko, 2019-09-23:
I"m seeing similar behavior.  My gyro pitch appears to be sampled at a much lower rate than gyro roll (35 ms).
[image]

closed 2019-12-06 by stale[bot]