iNavFlight/inav #4994: Multicopter RTH Behavior Changed In Version 2.2.1

KenImhof, 2019-08-05:
Not sure if this is a problem or new feature in version 2.2.1.

During multicopter RTH operation, if the position is manually changed via PITCH or ROLL input, the multicopter will stop and stay at is current location when the stick input is released.  It appears that the RTH transitions to POSHOLD, but the throttle stick altitude control has no effect.  If the multicopter is then manually flown back to the home position via stick commands, it will resume the RTH operation and land.

In previous versions, when the stick commands were released, the multicopter would automatically resume the RTH operation and return to the home position and land.

teckel12, 2019-08-05:
@KenImhof I have used RTH several times with 2.2.1 and it's quite common for me to make some corrections when in the final land descent. At least on the land descent, I can move the sticks and it doesn't hold. I use this to slightly change the home position (as I'm typically launching from a narrow strip of asphalt or concrete. I make a correction and it continues the RTH.

I don't think I've ever tried to move the sticks before it was in the land descent stage. What RTH method are you using? Also at what point are you moving the sticks? While it's returning or while landing?

KenImhof, 2019-08-05:
@teckel12 

Hi Tim, this only happens in the latest 2.2.x version.  I was going to look at fixing this issue, but I am starting to like this behavior.  As soon as RTH is initiated, if its position is altered while in-route to its home position, it will just stop at its current location as if it went into position/alt hold.  

Once it reaches is home position and begins its descent, everything works the same as the previous versions.  When descending, you can alter its position via the PITCH/ROLL input and it will update its landing home position.

Another nice RTH change I have noticed in this version is that you can change the YAW orientation while flying to its home position.

teckel12, 2019-08-05:
@KenImhof I'll play with this tomorrow.

hali9, 2019-08-05:
I confirm. When my quad performs RTH and I move sticks, he was stop like in poshold. Must have to turn off and re-enable RTH.
INAV 2.2.1 (a6d847482) OMNIBUSF4PRO

KenImhof, 2019-08-06:
Submitted Pull Request #4996. I left the YAW controllable during the RTH so it is possible to update the YAW or even pirouette during the RTH operation.

teckel12, 2019-08-06:
Played with this today and I have to admit I love it! Slight landing modifications seem reasonable. Drastic flight changes trigger POSHOLD. I cannot see a problem with this. If you let it RTH it will do so. If you want to take control, it will let you. The Yaw can totally be changed without RTH interpretation. Totally love it!

hali9, 2019-08-06:
While this behavior is desirable during RTH failsafe, it should not happen during normal RTH.
This should work as before, i.e. the possibility of correction yes, but RTH is only interrupted when the switch is turned off.

teckel12, 2019-08-06:
@hali9 I think what's happening is when there's a drastic stick interruption during RTH it detects that RTH is failing and falls back to POSHOLD.

I really think this is the only way RTH should work. If RTH isn't progressing as expected, fallback to POSHOLD. As RTH was triggered via a switch anyway, you can just re-engage RTH. I think this can avoid fly aways based on a non-calibrated compass or a compass with the wrong orientation.

KenImhof, 2019-08-06:
When multicopter gets stuck in this mode, I would call it suspended RTH mode, as it is not in the standard POSHLD mode.  In this mode, you cannot control the altitude and if you move it to the RTH destination position, it will resume the RTH operation and land.  I also have my quad in automatic low battery RTH and it is possible in the mode to get it stranded at the current altitude although I do have a low battery override option.

hali9, 2019-08-07:
@teckel12 When I move the sticks quad stops in "like poshold" but it is not poshold, he does not return to any poshold because before running RTH I was running angle and not poshold.
The question is whether this behavior was intended by programmers. I do not think so. 

If You want to interrupt RTH and enable poshold, you always have the option to use the switch. And the sticks should be used for correction.

> I think what's happening is when there's a drastic stick interruption during RTH

A small correction causes this

> As RTH was triggered via a switch anyway, you can just re-engage RTH

But I don't want reengage RTH on every time whenever I need to do a slight correction of the flight path, and at the moment I can't make any correction.

> I think this can avoid fly aways based on a non-calibrated compass or a compass with the wrong orientation.

Poshold also needs a compass sensor even more than RTH, so I don't understand how it can be safer.
I've had many cases when RTH worked and POSHOLD didn't. Or when my quad did the RTH and before the landing instead of turning it fly away and it was just causing wrong indication of the compass.
There is "sanity cheking" to prevent fly-away.

RTH should be trusted and do not make such circuses, because it is usually used to recover equipment after losing control, failure of fpv or loss of eye contact, etc. And in such cases, You can often even move the stick accidentally and this should not break the RTH.

digitalentity, 2019-08-07:
Fix merged

closed 2019-08-07 by digitalentity

neilm06, 2019-08-07:
This may be what is happening with RTH 2.2.x on the Matek 722 and Kakute F7. No one seemed to have an answer a month ago.