iNavFlight/inav #4997: beeper beeps ever boot with only RX_SET

laurentopia, 2019-08-06:
## Current Behavior
beeps on boot up

## Steps to Reproduce
beeper -all
beeper rx_set
save
reboot

## Expected behavior
no beep until beeper mode is on

## Additional context
https://pastebin.com/DtqHCYnD

# version
# INAV/MATEKF405 2.2.1 Jul  3 2019 / 22:31:06 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

eachine tyro 129 board

it works find in betaflight 4.0.3 with the matek f405 firmware

laurentopia, 2019-08-06:
it also beeper with `beeper -all`

digitalentity, 2019-08-07:
Initial beeper signal on plug-in is intentional, indicates successful boot up and can't be disabled

laurentopia, 2019-08-07:
It is a problem as it is piercing our eardrum. based on conversation with someone on the telegram chat, I'm not the only .
It would be better to have it off completely like in betaflight or a CLI `beeper BOOT` and let the user decide if they want to have BOOT sound or not.

teckel12, 2019-08-08:
@laurentopia Also solved by a small piece of tape over the speaker hole.

laurentopia, 2019-08-08:
> @laurentopia Also solved by a small piece of tape over the speaker hole.

 you’re implying that boot beep is necessary, thus implying that a beeper is necessary for normal bench operation of the FC

teckel12, 2019-08-08:
@laurentopia I'm just implying that it's easy to fix with a piece of tape.  That way you can leave all the beep controls as normal when on the bench.  Unless you don't want it to ever beep, in which case why not just disconnect the beeper?

Also, some FC's beeper only work when the battery is attached.  So a typical USB bench connection doesn't beep at all.

laurentopia, 2019-08-09:
That's a problem in this case because the buzzer's mouth is pointing towards the case to expose the disarm button.

The reason why I thought you were implying that is that forcing boot beep on us could be due to a use case where one wants to know if the FC is fried. I'm trying to understand why there is even an audible boot that's not controllable by a parameter, especially when MSP is on.

teckel12, 2019-08-10:
@laurentopia Not sure. Seems reasonable to not have any beeps. I have a few FC that have an inverted beeper, so they're full-on during the entire programming cycle. SO annoying! Which is why I've just resorted to the "tape fix" when on the bench. As there's no way to avoid it with sleeping family members.

digitalentity, 2019-08-10:
Just looked at the code. `beeper -SYSTEM_INIT` should disable beeping on boot.

laurentopia, 2019-08-13:
> I have a few FC that have an inverted beeper, so they're full-on during the entire programming cycle.

this crazy FC need [this programmable buzzer](https://us.banggood.com/Wholesale-Warehouse-FullSpeed-Lucky-Box-Beeper-Buzzer-5V-Built-in-Battery-for-RC-Drone-FPV-Racing-wp-Usa-1340935.html?rmmds=myorder)

> Just looked at the code. `beeper -SYSTEM_INIT` should disable beeping on boot.

I'll try that once it's back in one piece, if it works it's weird because I always do `beeper -all` `beeper RX_SET`

closed 2019-10-26 by stale[bot]