iNavFlight/inav #4998: Probleme with BTRX on jumper 16 2.23 - F4 f35 clone - inav 2.21 - Crossfire micro + nano and  inav script 1.71 on t16

dovichino, 2019-08-07:
Hi,

I have a problem with the btrx on my radio. On inav and in telemetry I see the voltage of my battery but impossible to have on the script inav. I start then maybe I made a bad adjustment?

[image]
[image]
[image]

[image]

wx4cb, 2019-08-07:
make sure you're running the development version, then delete and rediscover the sensors. there's documentation in the wiki to download and install the dev version.

dovichino, 2019-08-07:
Thanks for your help !
i have the last version 1.71. 
It's a new install, i bought my crossfire today. 
we can see the sensor work in telemetry,12.3v but dont work in the main panel.
I was delete and discover 5 times...
I do not understand.

wx4cb, 2019-08-07:
the latest DEV version is 1.7.2
https://github.com/iNavFlight/LuaTelemetry/wiki/Upgrade-to-Development-Build

dovichino, 2019-08-07:
ok thx i try that

dovichino, 2019-08-07:
I am in 1.72 and always the same problem after delete and discover the telemetry

teckel12, 2019-08-07:
@dovichino You should open an issue in the lua telemetry repo, not here in INAV. This is not at all related to INAV. Also, I'm the author of the Lua Telemetry script, so I'll personally be looking into this.

teckel12, 2019-08-07:
@dovichino Quickly looking into this, it seems you renamed the sensor. It should be `RxBt` but you renamed it `BtRx`. So that's why it can't find the voltage. Don't change sensor names.

If you're still having a problem, please open an issue in the correct repo (INAV Lua Telemetry).

dovichino, 2019-08-07:
A big thank you for your help!

I had not renamed RxBt to BtRx, it's the telemetry that says that name. I just rename it and it works except that if I delete and I discover again it is renamed BtRx and not RxBt. I am french can be a problem related to translation?

dovichino, 2019-08-07:
I just flashed with firmware 2.23 EN and I RxBt when I do discover.

So this comes from the firmware translation

digitalentity, 2019-08-07:
Wrong repository

closed 2019-08-07 by digitalentity