iNavFlight/inav #5000: [REQUEST] Manual (Emergency) Landing for Wings and Copters

b14ckyy, 2019-08-08:
## Current Behavior
Emergency landing is only triggered in case of sensor failure during Failsafe or as Landing at home point. 

## Desired Behavior
We should be able to trigger a GPS assisted landing with a switch

## Suggested Solution
Following Scenario: 

- A long range pilot realizes that the battery won't last for the way back
- pilot searches for a save landing spot like a open field and flies over it
- pilot triggers landing and plane will go into circle down with throttle cut to zero till it touches ground and disarms, Copters will fast descent at emergency landing descent speed
- maybe only uses throttle on a wing to compensate for strong wind drift above a specified threshold
- failsafe must be ignored in this state as in long range the pilot will likely loose connection before touchdown
- after landing, pilot can use telemetry data to find the plane in a very specific area

## Who does this impact? Who is this for?
All Long range pilots that touch the limit of their battery capacity. Also pilots that maybe have a faulty battery. 

## Additional context
This feature would be a huge security benefit as the pilot is able to control where the plane or copter will go down in a unpopulated area instead of letting it glide uncontrolled if the FC looses power too. A manual emergency landing won't be possible on long range as the connection will likely be lost at low altitudes and then the craft will likely failsafe and try to climb again. 

Setting Failsafe to LAND is no option on long range as the RTH should still work if there is no issue with the battery.

wx4cb, 2019-08-08:
silly question, but if they know it wont last all the way back and they search for a field etc, why can't they put the aircraft down manually ?

datamann, 2019-08-08:
This sounds like a good idea. If you try to take the plane down manually and looses radio connection the normal failsafe will trigger which is normally RTH and that would not be good thing if you are low on battery.

b14ckyy, 2019-08-11:
> This sounds like a good idea. If you try to take the plane down manually and looses radio connection the normal failsafe will trigger which is normally RTH and that would not be good thing if you are low on battery.

That's exactly the point.

Mateyhv, 2019-09-02:
Yes, that is a very desirable option. I suggest also improving the failsafe land feature since in its present state I don't think anyone use it in multicopters. Desarming the motors after a preset time just doesn't make sense. The copter might be still a hundred meters above ground or be spinning on the ground, winding grass or foreign objects wich surely will end in burning motors/escs. 
Copters must rely on the sudden bump on touchdown for desarming the motors, maybe .5secs after the high G-load. The same should be true for the manual emergency landing.

teckel12, 2019-09-02:
This sounds like a good idea. When you get low to the ground in your emergency landing location, you'll most likely loose connection at some point triggering RTH, which won't be a good thing.

b14ckyy, 2021-03-13:
oh just tumbled over this again. 
And I just had an idea how to trigger that mode without the need of a extra flight mode switch. 

the trigger could be, that the pilot goes to throttle zero FIRST and then hit the RTH switch in a certain amount of time. Like 2000ms. raisng the throttle will then go back into RTH. if RTH is engaged already, then zero throttle stick won't do anything. if the pilots glides for more than 2s and hits, rth, it is also triggered normally. this should prevent accidetal emergency landings and makes it easy and intuitive to abort it and resume to normal flight.

danarrib, 2021-03-13:
> Copters must rely on the sudden bump on touchdown for desarming the motors, maybe .5secs after the high G-load. The same should be true for the manual emergency landing.

Problem is that bumps caused by strong wind may lead to false-positives.

> the trigger could be, that the pilot goes to throttle zero FIRST and then hit the POSHOLD switch in a certain amount of time. Like 2000ms. raisng the throttle will then go back into POSHOLD. if POSHOLD is engaged already, then zero throttle stick won't do anything. if the pilots glides for more than 2s and hits, POSHOLD, it is also triggered normally. this should prevent accidetal emergency landings and makes it easy and intuitive to abort it and resume to normal flight.

Looks a bit too complicated for me. The idea of having a "Emergency Landing" Mode that can be attached to a channel range like any other mode is nice. And make it "failsafe-proof" like NAV-WP is the way to go.

b14ckyy, 2021-03-13:
> Problem is that bumps caused by strong wind may lead to false-positives.
>
>
> Looks a bit too complicated for me. The idea of having a "Emergency Landing" Mode that can be attached to a channel range [...]

The emergency landing is nothing the pilot will ever need to apply quickly in a hectic situation. This is also not intended to "land" the plane gently. It is more like a controlled crash in a safe area to not cause any casualties. 

If you run out of battery or got other problems like strong winds, Motor failure, prop loss, ESC failure, whatever could happen far out, this Emergency mode will just allow the pilot to start a controleld crash and make sure it does not go down on a street or in populated area. With some luck the plane will survive with minor damage. 

I don't see a reason why we should sacrifice a dedicated channel and switch for something that will hopefully never be needed. It is like a Airbag in the car. And the pilot should always have plenty of time to think about what to do in such situations. So a throttle low plus POSHOLD switch in a specific time frame would be a reasonable solution. And it must be able to easily abort by just raising the throttle or change mode.

breadoven, 2022-11-29:
I've got something working for this but the main issue is how you trigger it. Currently it uses PosHold + Throttle Low + Yaw Hi. This might work OK for a FW but it's not really suitable for a MR since it could be a legitimate, albeit slightly odd, PosHold manoeuvre (spinning descent). The easiest trigger would be a dedicated mode but this seems a bit of a waste for something that would rarely get used.

Once triggered the emergency landing can be cancelled by simply raising the Throttle and Failsafe is inhibited whilst it's active.

b14ckyy, 2022-11-30:
Oh i totally forgot about this request. 
Yes you are right this should not need a dedicated channel. Hence my suggestion of throttle low + POSHOLD within a certain time: https://github.com/iNavFlight/inav/issues/5000#issuecomment-798094598
this would also be recoverable by just applying throttle. 

Alternatively it could be coupled with poshold + disarm but then distance from home has to be taken into account as a threshold. In general this might be an even better way. Disarm should then just cut off the throttle but keep poshold with constant descent active at a specific descent rate the plane can achieve but should allow rearm in this state.

breadoven, 2022-11-30:
I thought this might be useful before when I was improving the emergency landing stuff. It's a simple change to do just using the existing Failsafe emergency landing code. It's just triggering it that's not obvious. Not sure about disarming, this affects a lot of stuff that's needed for a controlled emergency landing. I could just do a PR as is and leave it open to discussion regarding the trigger method.

breadoven, 2022-12-14:
I've now updated this to be triggered using only multi switching of PosHold (4 switches in 2s for now) similar to emergency Arming. It's now easily cancelled with throttle HIGH. This should work well for multirotors as well as fixed wing. Multirotor PosHold can be toggled with throttle mid rather than low as before which avoids a rapid descent when toggling the switch. Cancelling with throttle HIGH will cause a multirotor to shoot upward but only briefly so not really an issue. This is afterall intended as a last ditch attempt at a controlled landing anyway so better than nothing. I've also added position control during the emergency landing so it won't drift in wind.

b14ckyy, 2022-12-15:
this sounds fantastic. No waste of channels, no accidental trigger and easy to do as POSHOLD should be set on most planes and for sure anything for long range.

Jetrell, 2023-01-05:
@breadoven Any chance of adding this feature a Logic condition flight mode action too?

breadoven, 2023-01-06:
> @breadoven Any chance of adding this feature as a Logic condition flight mode action too?

Isn't that already covered by Flight -> is emergency landing ?

MrD-RC, 2023-01-06:
I'm sure Jetrell means to activate it.

Jetrell, 2023-01-06:
> 
> Isn't that already covered by Flight -> is emergency landing ?

Yes it is..  I just assumed that would be the conventional landing method? And not give the ability to activate this new approach?

b14ckyy, 2023-01-07:
If you add a trigger in Programming, then this must be done careful. Otherwise you will be stuck in EL and can't exit anymore. Especially if someone decides to use the LC with no loiter FM fitted to switches.

MrD-RC, 2023-01-07:
You shouldn't get stuck in it. If the LC is active, it should be active. If not, it shouldn't be. It would bypass loiter mode totally if operated by LC.

breadoven, 2023-01-07:
You could add a trigger to Programming but this would be the only one that would activate a flight mode (emergency landing isn't actually a flight mode as such). How would the Programming LC be triggered if it was used ?

I've got something I'm working on now that uses an OSD field to show warnings, GPS Fail etc, but it also provides functions. The warnings just pop up as they occur etc. The functions on the other hand work using a single mode (single channel required to operate). Short activation of the mode cycles though the available functions displaying them in the OSD field. The function displayed in the OSD field is triggered with a long activation of the mode. Ideally the mode should be activated using a momentary switch although it works with a non momentary also (just need to remember to deselect it). It's intended for rarely used "one shot" things such as manual emergency landing, emergency arming, Turtle mode ? etc, Not as easy to operate as a dedicated switch but better than toggling switches multiple time and more intuitive because the OSD field tells you which function is selected.

I was wondering how much of this can be done by Programming, some of it perhaps but it would be messy.

Jetrell, 2023-01-07:
> You could add a trigger to Programming but this would be the only one that would activate a flight mode (emergency landing isn't actually a flight mode as such). How would the Programming LC be triggered if it was used ?

I can think of a one case, of which I won't state here. Needless to say, options are always helpful, with the way FPV regulations in the world are going.

But besides that. I'm not a fan of multi-toggling a switch, to enable an operation. 
Logic conditions and timers, can be used to trigger a function some seconds after the switch it toggled, with an audio call, before it activates.
This way, you can always flick the switch off again, before the activation timer is complete. Thus, not accidentally turning it on.
I use this method for `VTX/camera power OFF`, `Home position reset` and `Log record disabled`.

breadoven, 2023-01-07:
Given this is really only intended to be used as a last resort I think the switch multi toggle should be fine and work for all users including those who don't use programmable radios, OpenTx etc (admittedly probably very few these days).

An LC could be added for this, it's just debateable if it's worthwhile for something that will rarely get used.

b14ckyy, 2023-01-08:
For LC in my option only a "normal" land mode would make sense that keeps Failsafe working. So the pilot has control until touchdown in case something happens. But a controlled crash mode should not be in LCs imho. 
keep in mind how many people think they can use Failsafe as a normal RTH mode in INAV.

closed 2023-01-27 by DzikuVx