iNavFlight/inav #5002: JHEMCU F7 - support !

yo8aiv, 2019-08-10:
Hi . i don't know if this is the right way to ask for new FC to be supported.

its called JHEMCU F7 target in BF is JHEF7DUAL. 
this board could be of better use in iNav than BF since it has so many uart's.
more info 
https://github.com/betaflight/betaflight/issues/8605
https://github.com/betaflight/betaflight/pull/7398

Thanks.

digitalentity, 2019-08-10:
@yo8aiv see https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

marianomd, 2019-08-11:
Hello @atgfpeyv could you kindly provide this FC to inav developers?

marianomd, 2019-08-30:
Hi, are there guidelines to port fc support from betaflight to inav? Is it too different? I'm considering buying it and maybe I can help. Some guides to get up and running a development environment/sdk/ide? Thanks

teckel12, 2019-08-30:
@marianomd Best to start with a similar F7 dual FC target in INAV then use the Betaflight taeget for the FC to add to make the appropriate changes. Then testing each UART, I2C, buzzer, LED, etc. (which is why having the FC in-hand is really required).

Anyway look in Betaflight for a similar FC and start there.

marianomd, 2019-08-30:
Thank you Tim, great advise!

waynoedm, 2019-09-10:
dump

# version
# Betaflight / JHEF7DUAL (JH7D) 4.0.6 Sep  1 2019 / 00:43:08 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name JHEF7DUAL
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 C15
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B04
resource MOTOR 4 B03
resource MOTOR 5 C09
resource MOTOR 6 C08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource I2C_SCL 1 B06
resource I2C_SCL 2 NONE
resource I2C_SCL 3 NONE
resource I2C_SCL 4 NONE
resource I2C_SDA 1 B07
resource I2C_SDA 2 NONE
resource I2C_SDA 3 NONE
resource I2C_SDA 4 NONE
resource LED 1 A15
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_SCK 4 NONE
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MISO 4 NONE
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource SPI_MOSI 4 NONE
resource CAMERA_CONTROL 1 B08
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C00
resource ADC_CURR 1 C01
resource ADC_EXT 1 NONE
resource BARO_CS 1 NONE
resource PINIO 1 C14
resource PINIO 2 B09
resource PINIO 3 NONE
resource PINIO 4 NONE
resource USB_MSC_PIN 1 NONE
resource FLASH_CS 1 C13
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_EXTI 2 C03
resource GYRO_CS 1 B02
resource GYRO_CS 2 A04

# timer
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B03 AF1
# pin B03: TIM2 CH2 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer C08 AF3
# pin C08: TIM8 CH3 (AF3)
timer A08 AF1
# pin A08: TIM1 CH1 (AF1)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)

# dma
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_TX 4 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma SPI_RX 4 NONE
dma ADC 1 NONE
dma ADC 2 NONE
dma ADC 3 0
# ADC 3: DMA2 Stream 0 Channel 2
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin A03 NONE
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin B04 0
# pin B04: DMA1 Stream 4 Channel 5
dma pin B03 0
# pin B03: DMA1 Stream 6 Channel 3
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2

atgfpeyv, 2019-10-27:
Forgive me for being late, so what should I do for inav to support this board?

craterghost, 2019-11-01:
I don't get it either. Is it possible to change the JHEF7DUAL.hex file from betaflight to one that's readable by INAV? What can we do to speed up the process of Supporting the FC?

xD90o1, 2019-11-01:
> I don't get it either. Is it possible to change the JHEF7DUAL.hex file from betaflight to one that's readable by INAV? What can we do to speed up the process of Supporting the FC?

Looking at the target.c of both betaflight and inav targets they seem to differ quite  alot although I somehow managed to make a hex file for this specific fc though. I still couldn't manage to make the user modes work sadly

craterghost, 2019-11-01:
would u mind sending me the hex file u already made?

xD90o1, 2019-11-01:
> Forgive me for being late, so what should I do for inav to support this board?

Also as per the New Hardware Policy md, New targets are accepted into INAV code if the board manufacturer itself provides the specs, schematics, and production samples to at least 2 core devs. Or, a community member, or board manufacturer provides board samples to at least 1 core dev and the target itself already exists in the official Cleanflight or Betaflight firmware.

craterghost, 2019-11-01:
or I just write the code myself... can't be that hard tbh... not worth spending 20 bucks. The specs are pretty much the same as the matek F722 so I just try modding the code for the matek board with the parameters from the jhef7dual code from betaflight.

redeyejoe, 2019-11-13:
Has anyone had any luck getting this f7 flight controller working with i nav??
I would like to be able to use its features with more gps control...

xD90o1, 2019-11-20:
> Has anyone had any luck getting this f7 flight controller working with i nav??
> I would like to be able to use its features with more gps control...

I have quite some success except for the user modes. I just can't seem to get it working

aknack, 2019-12-30:
hi inav team! great so so far!

is there any progress on this FC?

JoseAntonioMG, 2020-01-09:
Hello, I also have this FC and I have configured it with Betaflight, but I would like to have the INAV firmware, since I am building a Long Range quad with GPS.

redeyejoe, 2020-01-10:
> > Has anyone had any luck getting this f7 flight controller working with i nav??
> > I would like to be able to use its features with more gps control...
> 
> I have quite some success except for the user modes. I just can't seem to get it working

did you make you own fireware or what are you using to make it work partially?

xD90o1, 2020-01-10:
I started with a source code of a dual flight controller that's present
here in inav and then I just go ahead and change the pinouts accordingly to
match the ones used in the JHE F7 Dual. It's already working but I just
can't make the user modes work

redeyejoe, 2020-01-10:
> I started with a source code of a dual flight controller that's present here in inav and then I just go ahead and change the pinouts accordingly to match the ones used in the JHE F7 Dual. It's already working but I just can't make the user modes work
> […](#)
Can you share you file?

smaller09, 2020-01-14:
> I started with a source code of a dual flight controller that's present here in inav and then I just go ahead and change the pinouts accordingly to match the ones used in the JHE F7 Dual. It's already working but I just can't make the user modes work
> […](#)

did you add the pinio pins into target.h
#define USE_PINIO
#define USE_PINIOBOX
#define PINIO1_PIN                  PC14 // VTX power switcher
#define PINIO2_PIN                  PB09 // 2xCamera switcher

soninua, 2020-01-30:
Hello!
Are there any changes with this board? Someone was able to do it with INAV?

redeyejoe, 2020-01-30:
> > I started with a source code of a dual flight controller that's present here in inav and then I just go ahead and change the pinouts accordingly to match the ones used in the JHE F7 Dual. It's already working but I just can't make the user modes work
> > […](#)
> 
> did you add the pinio pins into target.h
> #define USE_PINIO
> #define USE_PINIOBOX
> #define PINIO1_PIN PC14 // VTX power switcher
> #define PINIO2_PIN PB09 // 2xCamera switcher

Is there any way you could just post your working bin file so I can try it how it is? Thanks

smaller09, 2020-02-03:
> > > I started with a source code of a dual flight controller that's present here in inav and then I just go ahead and change the pinouts accordingly to match the ones used in the JHE F7 Dual. It's already working but I just can't make the user modes work
> > > […](#)
> > 
> > 
> > did you add the pinio pins into target.h
> > #define USE_PINIO
> > #define USE_PINIOBOX
> > #define PINIO1_PIN PC14 // VTX power switcher
> > #define PINIO2_PIN PB09 // 2xCamera switcher
> 
> Is there any way you could just post your working bin file so I can try it how it is? Thanks

[inav_2.4.0_JHEMCUF7DUAL.zip](https://github.com/iNavFlight/inav/files/4148858/inav_2.4.0_JHEMCUF7DUAL.zip)
not have time to test yet, please report back

aknack, 2020-02-04:
the .hex included in the zip does not work on my FC.
after flashing and powering it makes 4 (or 5) very quick beeps, followed by 2 longer ones.
this repeats until the FC reboots on itself. and then beeps again in same manner.

after reflashing it with
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.3 Jan 16 2020 / 11:35:53 (543a5e1ba) MSP API: 1.42
# manufacturer_id: JHEF   board_name: JHEF7DUAL   custom defaults: YES
```

all is fine again.

xD90o1, 2020-02-04:
> > > > I started with a source code of a dual flight controller that's present here in inav and then I just go ahead and change the pinouts accordingly to match the ones used in the JHE F7 Dual. It's already working but I just can't make the user modes work
> > > > […](#)
> > > 
> > > 
> > > did you add the pinio pins into target.h
> > > #define USE_PINIO
> > > #define USE_PINIOBOX
> > > #define PINIO1_PIN PC14 // VTX power switcher
> > > #define PINIO2_PIN PB09 // 2xCamera switcher
> > 
> > 
> > Is there any way you could just post your working bin file so I can try it how it is? Thanks
> 
> [inav_2.4.0_JHEMCUF7DUAL.zip](https://github.com/iNavFlight/inav/files/4148858/inav_2.4.0_JHEMCUF7DUAL.zip)
> not have time to test yet, please report back

[JHEF7DUAL.zip](https://github.com/iNavFlight/inav/files/4153482/JHEF7DUAL.zip)

Here's my try at the code for the JHE F7 Dual

soninua, 2020-02-04:
> > > > Я начал с исходного кода контроллера двойного полета, который присутствует здесь в inav, а затем я просто иду вперед и изменяю pinouts соответственно, чтобы соответствовать тем, которые используются в Jhe F7 Dual. Это уже работает, но я просто не могу заставить пользовательские режимы работать 
> > > > [...](#)
> > > 
> > > 
> > > вы добавили булавки pinio в target?h
> > > #определите USE_PINIO
> > > #определите USE_PINIOBOX
> > > #определить PINIO1_PIN PC14 / / VTX Power switcher
> > > #определить pinio2_pin pb09 / / 2xcamera switcher
> > 
> > 
> > Is there any way you could just post your working bin file so I can try it how it is? Thanks
> 
> [inav_2.4.0_JHEMCUF7DUAL.zip](https://github.com/iNavFlight/inav/files/4148858/inav_2.4.0_JHEMCUF7DUAL.zip)
> not have time to test yet, please report back

Hello!
This version does not work, there is no communication in the configurator, the board reboots after 5-6 seconds.

soninua, 2020-02-04:
> > > > > Я начал с исходного кода контроллера двойного полета, который присутствует здесь в inav, а затем я просто иду вперед и изменяю pinouts соответственно, чтобы соответствовать тем, которые используются в Jhe F7 Dual. Это уже работает, но я просто не могу заставить пользовательские режимы работать 
> > > > > [...](#)
> > > > 
> > > > 
> > > > вы добавили булавки pinio в target?h
> > > > #определите USE_PINIO
> > > > #определите USE_PINIOBOX
> > > > #определить PINIO1_PIN PC14 / / VTX Power switcher
> > > > #определить pinio2_pin pb09 / / 2xcamera switcher
> > > 
> > > 
> > > Есть ли какой-либо способ, которым вы можете просто опубликовать свой рабочий файл bin, чтобы я мог попробовать его, как это есть? Спасибо
> > 
> > 
> > [inav_2. 4.0_JHEMCUF7DUAL.zip](https://github.com/iNavFlight/inav/files/4148858/inav_2.4.0_JHEMCUF7DUAL.zip)
> > еще не успел протестировать, пожалуйста доложите
> 
> [JHEF7DUAL.застежка-молния](https://github.com/iNavFlight/inav/files/4153482/JHEF7DUAL.zip)
> 
> Вот моя попытка в коде для Jhe F7 Dual

Hello!
This version works, the configurator 2.2.1 works fine. I did not have time to test everything. There is communication with the board in the configurator program, the sensors are working. We need to test further. can this version be rebuilt under 2.4.0 INAV?

xD90o1, 2020-02-05:
It can be. I think the target source codes for the 2.2.1 and 2.4.0 don't
differ that much.

aknack, 2020-02-06:
it builds unter origin/master:
```
Building JHEF7DUAL succeeded.
ak@osboxes:inav$ ll obj/inav_2.3.0_JHEF7DUAL.hex
-rw-r--r-- 1 ak ak 897K Feb  6 03:22 obj/inav_2.3.0_JHEF7DUAL.hex

```
and also builds unter 2.4.0-RC2
```

Building JHEF7DUAL succeeded.
ak@osboxes:inav$ ll obj/inav_2.4.0_JHEF7DUAL.hex
-rw-r--r-- 1 ak ak 924K Feb  6 03:30 obj/inav_2.4.0_JHEF7DUAL.hex

```
[inav_2.4.0_JHEF7DUAL.hex.zip](https://github.com/iNavFlight/inav/files/4163984/inav_2.4.0_JHEF7DUAL.hex.zip)

xD90o1, 2020-02-06:
I want to know if the user modes work out for you guys

aknack, 2020-02-06:
user modes are NOT morking for me.
according to  https://ae01.alicdn.com/kf/HTB11y.9VMHqK1RjSZFgq6y7JXXaT.jpg
i can power the vtx with BF413 , 
but not with the compiled .hex unter INAV.  this version connects with the configurator.

xD90o1, 2020-02-06:
Unfortunately, that's the brick wall I've ran into months ago when I use
that firmware which I tried coding for that fc for inav.

aknack, 2020-02-06:
a shame, since its a good F7+baro for a very good price.

smaller09, 2020-02-07:
> user modes are NOT morking for me.
> according to https://ae01.alicdn.com/kf/HTB11y.9VMHqK1RjSZFgq6y7JXXaT.jpg
> i can power the vtx with BF413 ,
> but not with the compiled .hex unter INAV. anyhow this version connects with the configurator.
I think the problem should be in the config.c

```
void targetConfiguration(void)
{
pinioBoxConfigMutable()->permanentId[0] = 40;
pinioBoxConfigMutable()->permanentId[1] = 41;
}

#endif

```
the two ID should be the MENU ID, so it could not use the number from BF
I took that from the MATEKF765

void targetConfiguration(void)
{
    pinioBoxConfigMutable()->permanentId[0] = 47;
    pinioBoxConfigMutable()->permanentId[1] = 48;
}

You can try it.

aknack, 2020-02-08:
**smaller09** , you''re the man ! :)

i can confirm user-modes are working. can't confirm yet, that it will fly.
[inav_2.4.0_JHEF7DUAL_smaller09_menu_patch.zip](https://github.com/iNavFlight/inav/files/4174416/inav_2.4.0_JHEF7DUAL_smaller09_menu_patch.zip)

xD90o1, 2020-02-08:
As long as there are no conflicts with the timers, it'll fly just fine

aknack, 2020-02-13:
all seems fine, but i'm unable to arm. still debugging. 
any ideas?

aknack, 2020-02-13:
Pre-arming checks : all green
cli status : 
`Arming disabled flags: CLI
`
gps shows 6 satellites : ready to fly
T16 bound via fport : ok

still it won't arm if detached from computer.

xD90o1, 2020-02-18:
> Pre-arming checks : all green
> cli status :
> `Arming disabled flags: CLI `
> gps shows 6 satellites : ready to fly
> T16 bound via fport : ok
> 
> still it won't arm if detached from computer.
> [JHEF7DUAL2.4.0_dump_all.zip](https://github.com/iNavFlight/inav/files/4198392/JHEF7DUAL2.4.0_dump_all.zip)

Do you have a beeper? Beepers really help a lot when knowing if it's ready to arm.

aknack, 2020-02-20:
yes. i have a beeper.
opentx-amber-sound-pack tells me "readly to fly"
the beeper stays silent when arming, but the FW does not arm.

michaelj0hn, 2020-02-23:
As I have the same FC and was surprised there wasn't an INAV image for it, I had to google :) Found this discussion, used the image that was posted here. I can confirm that the quad arms and flies without any problems.

JoseAntonioMG, 2020-02-23:
Hello, I have installed the firmware that has been published here and, apparently, the quad works, it is armed and it seems to fly, the yaw does work and I can turn it, the throttle also works and I can go up and down, but when trying to use the pitch and roll, the quad does not respond and remains in the same position.

The quad uses GPS and magnetometer, I have calibrated it correctly (gyroscope, accelerometer and compass).

xD90o1, 2020-02-24:
> Hello, I have installed the firmware that has been published here and, apparently, the quad works, it is armed and it seems to fly, the yaw does work and I can turn it, the throttle also works and I can go up and down, but when trying to use the pitch and roll, the quad does not respond and remains in the same position.
> 
> The quad uses GPS and magnetometer, I have calibrated it correctly (gyroscope, accelerometer and compass).
> 
> I have changed the firmware for the betaflight and everything works perfectly.

That's odd..
I've changed some of its timers since it conflicts with existing timers when I jumped the gap to iNav but then I seem to overlook the issue and focus only on fixed wing aircrafts that I forgot to check if it works for quads as well.

michaelj0hn, 2020-02-24:
> [..] when trying to use the pitch and roll, the quad does not respond and remains in the same position.

I do not have any issues like this, quad is moving well in all directions.

> The quad uses GPS and magnetometer, I have calibrated it correctly (gyroscope, accelerometer and compass).

I do not have GPS or a magnetometer connected, but assuming this should not make a difference when in flying in ANGL mode.

michaelj0hn, 2020-02-25:
When testing ALTHOLD, it however resulted in the quad violently flipping over and crashing to the ground. This was with pure ANGL mode, no compass or GPS modes had been added. Any idea what this could have caused?

xD90o1, 2020-02-27:
Just to let you know I've forked this repo with the JHE F7 Dual target added so I can test more on this.

aknack, 2020-02-28:
@xD90o1 
im building a new quad based on this FC [ pro version with baro ] and i will be using inav_2.4.0_JHEF7DUAL.hex from your fork.

xD90o1, 2020-02-28:
I haven't tested this yet on the deluxe version as I only have the acro version but hopefully they would just be the same since acro lacks only a second IMU and a Flash Memory

aknack, 2020-02-28:
[image]

step 1: r9 mini on t2 via fport: works.

aknack, 2020-02-28:
[image]
step 2: attached ESC, flahed them successfully via jesc. -> passthrough works fine.

aknack, 2020-02-28:
[image]

step 3: attached motor on M1. ESC initialisation peep suggests: all fine.

aknack, 2020-02-28:
step 4: rc-mode: angle.
arming, and look whats happening:
https://youtu.be/MdWsASn1UHs

aknack, 2020-02-28:
ok, that video might help you guys for debugging:

https://youtu.be/GXO2XZ3cnbc

xD90o1, 2020-02-28:
It might be one of two problems which is

A: ESC Issue - when you arm, the motor spins for awhile and stops even when armed, yeah? That could probably mean a bad ESC or;

B: Timer Conflict - Although I've already made sure that no such conflicts would appear in config.c

xD90o1, 2020-02-28:
I haven't really tested it yet with DShot with a Multirotor config since the ESC I had used in my plane is a generic hobbywing 30A that can only use PWM Signal.

aknack, 2020-02-28:
*LOL* im ernst ? ok. i can support you with hardware if you whish.

xD90o1, 2020-02-28:
That's the deluxe board you're using, right? Mine is the acrk versuon with a blue color on the solder mask of its board

aknack, 2020-02-28:
jepp.

there is the barometer:

[image]

xD90o1, 2020-02-28:
From what I've read however in the website, the difference between acro and
deluxe is that the deluxe has dual gyro and a flash memory while the acro
doesn't have those

aknack, 2020-02-29:
i tried to switch the gyro
```
set gyro_to_use=1
set acc_hardware = ICM20689
save
```
the FC now seems to reboot over and over.
i'm **not** able to connect to the FC with inav

[144 more replies]

closed 2020-05-17 by digitalentity