iNavFlight/inav #5007: Diagnostic CLI command for i2c

laurentopia, 2019-08-13:
## Current Behavior
I2C is a black box, a very very black box and a mag may not work for unknown reason

## Desired Behavior
some clarity, at least a list of i2c peripherals

## Suggested Solution
Add CLI command to list i2c peripheral addresses, it would help a lot

## Who does this impact? Who is this for?
bn880 users, some times auto mag doesn't work

digitalentity, 2019-08-13:
I think figuring out I2C device addresses is out of scope for INAV since we can't allow arbitrary addresses anyway to avoid odd behavior. Most of I2C issues are hardware-related (mis-wiring, lack of pull-up resistors etc).

laurentopia, 2019-08-19:
This is not about allowing arbitrary addresses, it's about **listing** addresses of everything on the i2c bus to help diagnostic.

closed 2020-12-11 by DzikuVx