iNavFlight/inav #5008: Mode ranges with logical operators

poedk, 2019-08-13:
## Current Behavior
At the moment, programming modes is not very intuitive (to me).
You have to program the receiver to handle switch combinations (two 3 point switches = nine combinations). 

## Desired Behavior
I woul dlike to be able to do all the programming in INAV.

## Suggested Solution
If you could add logical operators (at least AND/OR) to range combinations. You wouldnt have to program the receiver.
That way you could define a range like: Ch8(1300 to 1700) AND CH9(900 to 1300) where the mode is only selected when CH8 is in the middle and CH9 is down.
If NOT AND and NOT OR are added too, it would be possible to set up a rule, that i.e. only allows AUTOTUNE in MANUAL mode.

## Who does this impact? Who is this for?
All users.

## Additional context

Pairan, 2019-08-13:
hmmm ... My Taranis "speaks" some of the modes and settings, changes "flightmode" in the display. This won't be possible if it is only programmed in iNav. So I'll have to programm this any way ...

poedk, 2019-08-13:
I agree. There will always be stuff that can only be done in the transmitter. But in my opinion my proposal would improve the Modes setup while reducing the need for receiver programming.