iNavFlight/inav #5009: Altitude Hold Issue

mattschinkel, 2019-08-13:
Hi, I've been trying to setup altitude hold. My quad-copter jumps right after enabling alt hold. I set nav_mc_hover_thr to my hover throttle, and I think I have everything else set ok.

I tried to troubleshoot in blackbox explorer. I notice navTgtPos[2] seems incorrect (staying steady at -24.95m) with my drone starting alt-hold at 1.3m. I believe my baro is working correctly and I do have foam covering it.

In the attached blackbox image, I enabled alt-hold for 3 seconds. in the middle of that you see the motors spin up full, and over the next few seconds my drone has so much momentum that it's way up in the sky after I turn alt hold off.

I'm using inav_2.2.1 OMNIBUSF4V3_S5S6_SS

Please take a look, thanks.
Matt.

[image]
[CLI_settings_iNav_Aug_12_2019.txt](https://github.com/iNavFlight/inav/files/3498722/CLI_settings_iNav_Aug_12_2019.txt)
[LOG00008.TXT](https://github.com/iNavFlight/inav/files/3498723/LOG00008.TXT)

teckel12, 2019-08-14:
@mattschinkel This problem is typically caused by vibration. Could be lack of soft-mounting for flight controller or misconfigured PIDs.

Alt hold with the same hardware and version you are running work fine for me.

mattschinkel, 2019-08-14:
Should navTgtPos[2] become the value of baro meters when alt-hold is switched on? Could someone please take a look at their flight logs?

jiad78, 2019-08-16:
I think that I have similar issue at matek 722 std (instaled at original vibration dumpers and baro coverd by foam). During last flight I focused at different things but my drone also gone rocket 2 times and I saw "-" value of altitude readings.

gilglecio, 2019-09-09:
i use inav 2.2.1 and matek f405 ctr controller, ALTHOLD works perfectly, but after i fly in ACRO, when i switch to HOLD, the center of the stick becomes the 1750 position, ie with the stick in the center, the drone If it falls towards the ground, it is only locked at height if the stick is at 75%.

LawrenceLDJr, 2019-09-14:
This happened to me using DShot 600, and once on multishot. The DShot event I had alt hold enabled the full range of 3 pos switch. Max/min throttle til I was near ground and disarm. Multishot event I disabled alt hold and all was well. INav has a learning curve, and doesn't like my beat up year old betaflight quads. Which when I reflash to betaflight fly well still. The analog ESC protocols must be calibrated.

mattschinkel, 2019-09-14:
I reset all my settings and started over, it's fine now. I have no idea what the issue was.

closed 2022-11-26 by breadoven