iNavFlight/inav #5010: AHI/Crosshair offset to the right 2.2.1

Jetrell, 2019-08-14:
I have tried it with 2 different FC's and 3 different camera's in PAL. And the result is the same.
I went looking through the CLI and could only found  `osd_horizon_offset`. Which moved it up and down, not left or right.
Is there a command I have missed?

The preview is correct.
[image]

While the OSD has the cross hair offset to the right by one character.
[image]

FinalFrag, 2019-08-14:
This seems like duplicate of #4911

digitalentity, 2019-08-14:
Duplicate of #4911

closed 2019-08-14 by digitalentity