iNavFlight/inav #5012: Picture in picture osd

samcollins910, 2019-08-14:
## Current Behavior
Single view fpv

## Desired Behavior
Would it be possible, with the advent of FC,s that can switch between two fpv camera feeds to have a picture in picture in order to see the main fpv image and have inserted, possibly on a switch, a second view in order to see either a different perspective or view, ie a rear ward. 

## Suggested Solution
I have no suggested solutions to this. 

## Who does this impact? Who is this for?
For all fpv pilots but probably more interest to the long range flyers. 

## Additional context
Such a system would provide for greater spacial awareness when flying fpv. Making the act safer for both the pilot and other users of rhe same airspace. An fpv rear view mirror as it were.

teckel12, 2019-08-14:
@samcollins910 Not possible with current OSD hardware. Out of scope for INAV till the hardware exists to do this.

DzikuVx, 2019-08-14:
I think we can close it as unrealistic with the current generation of flight controllers

closed 2019-08-14 by DzikuVx