iNavFlight/inav #5014: ESC/Motor1 Output not working on MatekF722SE

SteffenFPV, 2019-08-16:
## Current Behavior
Multicopter/Quad Configuration.
After connection of Battery, ESC for motor1 recognizing no data.
In armed mode motor1 is not runnning.
Independent of Protocol (PWM, OneShot, DShot)
INAV V2.2.1 and older.
Board = MatekF722SE
ESCs = KISS ESC25A

## Steps to Reproduce
Config INAV for Multicopter Quad.
Connect Battery and arm motors.

## Expected behavior
All ESCs detect protcol and all motors are running after arming.

## Suggested solution(s)

## Additional context
Hardware (ESC & MatekF722SE) is complete replaced -> no change.
If the Matekf722SE is initial powered by USB only and the Battery is connected after this, then all ESCs working and all motors running after arming.
 
# version
# INAV/MATEKF722SE 2.2.1 Jul  3 2019 / 22:44:40 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

teckel12, 2019-08-16:
@SteffenFPV I'm running two Matek F722SE quads with 2.2.1 and neither have this problem. What happens if you switch the motors or the wires so different motors are driven. What happens if you control the motors via Blheli? Sounds like maybe you didn't go through proper diagnostic steps? 

I can tell you it's not INAV. Must be your FC, ESC, motors, wiring, INAV settings, etc. Try starting with clean INAV settings, set the mix and test the motors in the configurator (I believe that's all you need to do to test the motors). Anyway, the point is to keep it simple and reduce any possible config issues from being at fault. Basically, start from the basics like testing in Blheli and the configurator.

SteffenFPV, 2019-08-18:
@tecke12 thank you, I switched the wires, and ESCs, everytime Motor1 fails. I'm shure that there is no HW Problem, because if USB is connected, everything works fine. Also in INAV configurator>motor test, all motors running perfectly. For testing I flashed betaflight on this board, there is no problem, all motors working, also whitout connected USB.

teckel12, 2019-08-18:
@SteffenFPV I'd still go the basics route. Start with no programming, set the mix, receiver and arm switch (only using the configurator, no CLI). Then try arming. Basically, avoiding any possibility of a setting issue.

To me, because it works when USB is connected, it sounds more like a hardware problem. INAV doesn't do something different when USB is connected.

Finally, when you say motor 1 doesn't spin, do you mean the same motor? In other words, if you switch the wires around (say switch the motor 1 and 2 on the flight controller) would then it be motor 2 on the quad doesn't spin, or is it the same physical motor that doesn't spin?

ykyandy, 2019-08-25:
My f4v3 has a similar problem, but when I connect the battery and plug in the usb, even the disarm motor No. 1 will turn quickly without warning, scared me to dare to test the paddle now.  It seems that the motor does not start automatically when the usb is not inserted.

teckel12, 2019-08-25:
@ykyandy Try using multishot, zero issues ever with multishot. With Oneshot and DSHOT I've had occasional issues, so I totally avoid those.

SteffenFPV, 2019-08-25:
@teckel12, I flashed inav new with full chip erase and restore settings to default. After that I set the mixer to multirotor, quad X and enable the ESCs and set DShot 300.
After Power up the ESCs 2...4 "talking" via the motors "...DShot detected, ready" But ESC1 detects no protocol and gets not ready. (ESC1=Motor Output S1).
If the FC is powered up before with USB then everything works fine.
Maybe this is a problem with the interaction between inav and the KISS ESCs?

teckel12, 2019-08-25:
@SteffenFPV Don't use DSHOT, only multishot IMHO.

ykyandy, 2019-08-26:
> @ykyandy尝试使用multis
Sorry, it is my personal problem.  I chose the wrong model!  Choose to be a helicopter.  In fact, I am a multi-rotor.

> @ykyandy Try using multishot, zero issues ever with multishot. With Oneshot and DSHOT I've had occasional issues, so I totally avoid those.

Sorry, it is my personal problem.  I chose the wrong model!  Choose to be a helicopter.  In fact, I am a multi-rotor.

DzikuVx, 2019-08-28:
I have 2 F722-SE running INAV 2.2.1 in MR configuration and it's working without any problems. Looks like hardware issue

closed 2019-08-28 by DzikuVx