iNavFlight/inav #5018: f35 lightning update to 2.2.1

OzBlueDragon, 2019-08-17:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)
* [How to setup INAV on a flying wing](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLn_lCX7UB2OPLzR4w5G4el3)
* [How to setup INAV on a 5" quad](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLl9M6Vm-ZNLMCr6_9yNVHQG)

OzBlueDragon, 2019-08-17:
I'm trying to update my F35 Lightning to 2.2.1 but it won't update!!
It keeps saying "No response from the bootloader, programming: FAILED" !!!
This is beyond frustrating as I've never had this problem before on any other FC!
It wouldn't connect to my PC even after updating every Driver and everything so I switched to my iMac and it connects but won't do the update.
Please can someone help before I lose my Sh1t???
TIA.

DerElch87, 2019-08-17:
Please try the driver fix from Impulse RC:
https://impulserc.com/pages/downloads
If you need help how to use this tool, please google for the usecase.

OzBlueDragon, 2019-08-17:
> 
> 
> Please try the driver fix from Impulse RC:
> https://impulserc.com/pages/downloads
> If you need help how to use this tool, please google for the usecase.

Thanks friend trying it now on my brothers PC.

OzBlueDragon, 2019-08-17:
Ok found the problem! I didn't listen to my brother to run an earlier version of iNav 2.0.0
It has now connected on my iMac so lets see if I can update it now.

OzBlueDragon, 2019-08-17:
OK I've downloaded the iNav 2.1.4 Configurator and it will not let me flash the 2.1.0 Firmware?
why won't it update the firmware???
This is driving me nuts!!!

DerElch87, 2019-08-18:
Nice.
If everything is OK now, please close the issue.
BR

OzBlueDragon, 2019-08-18:
Yep ok still learning how to use this platform. 
The other issue is 2.2.1 won’t open on my Windows Laptop running 8.1 so I have to use the Chrome App Configurator but it works on my brothers laptop in Windows 10. 
Any help with this or should I search or start a new thread?

teckel12, 2019-08-18:
@OzBlueDragon You should upgrade from windows 8.1 to 10 IMHO. Windows 8.x was a mistake and no one should be using it.

OzBlueDragon, 2019-08-20:
@teckel12 yes mate that’s my next plan. Lucky my brother is in IT so I’ll get him to upgrade it soon.

closed 2019-08-20 by OzBlueDragon