iNavFlight/inav #5021: Crossfire Nano Diversity Receiver and Full Size Crossfire Transmitter not working with Inav 2.2.1 on Matek F411 Wing when updating Crossfire Transmitter to from 2.93 to 2.95

JWWFPV, 2019-08-17:
## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

teckel12, 2019-08-17:
@JWWFPV Duplicate #4855

In the future, when opening an issue, please use the subject field and body appropriately, keeping the subject concise and filling out the body information for your issue details.

Also, first searching for currently opened issues as to not create duplicate issues for the same problem (Smart Audio 2.1 not working in INAV). Currently, you'll need to manually select the frequency and power.

teckel12, 2019-08-17:
@JWWFPV Try Crossfire v3.0, I'm running it without any issues.

JWWFPV, 2019-08-17:
I will follow proper procedure on next post.  Thanks.  

 

I am using the TBS Agent to update my crossfire transmitter and it only shows 2.95.  Any tip on how to get 3.0?

 

 

John W. Wade II

5196 Candlewood Dr.

Fayetteville, NY  13066

(315) 569-6194 Cell

JWWFPV, 2019-08-17:
I see it now….  I did not have Beta versions available.  I am updating now and will report if that fixes my issue.

 

Thanks.

 

John W. Wade II

5196 Candlewood Dr.

Fayetteville, NY  13066

(315) 569-6194 Cell

JWWFPV, 2019-08-17:
Update to 3.0 corrected issue.

 

Thanks

 

 

John W. Wade II

5196 Candlewood Dr.

Fayetteville, NY  13066

(315) 569-6194 Cell

JWWFPV, 2019-08-17:
Update to version 3.0 corrected issue.

Thanks.

closed 2019-08-17 by JWWFPV