iNavFlight/inav #5022: Adding activation of internal pullups for i2c to the cli

KilianE, 2019-08-17:
## Current Behavior
Everytime you have to use the i2c, you have to add external resistors. Not on every device but on some. Sometimes this is hard, because you dont have enaugh space, or you dont have any resistors.

## Desired Behavior
You can activate the two internal pullup resistors in the CLI.

## Suggested Solution
Add a command to the CLI, which activates the the pullup resistor on the given pin
Like:
set pullup_on_pin.. = ON/OFF
Or more easy:
set i2c_pullup = ON/OFF

## Who does this impact? Who is this for?
For everyone who wants to use the i2c bus, without adding external resistors. Like for a barometer, compass, etc...

## Comment:
I am using the OMNINXTF7 (OMNIBUSF7NXT) so maybe someone can add the few lines of code, give me a hex and I will test this feature. If it works, maybe you can add it to inav completely.

teckel12, 2019-08-17:
@KilianE I haven't used the internal pull-up resistors on the F4/F7 but on the ATmega328 (Arduino) the internal pull-up resistors don't always work. For some things I use the internal pull-up resistors, but for i2c I've always used external resistors.

I like the idea, just not sure how successful it will be.

KilianE, 2019-08-17:
Thats the reason, why I said, that I can test it and doing also some research... I really like to try this feature, so maybe you can give me a trial version? I will buy resistors, but I have to wait till they will arrive... It would be really cool if I could try it, and I dont think that it needs that much time to add this little feature...

rrk1, 2019-08-18:
Isn't internal pull-up already enabled by INAV's driver when UART is initialized to I2C mode?

Anyway, very easy to verify even without oscilloscope - if you short any of SDA or SCL to the ground via multimeter set to milliamp range, a current of about 1mA will flow when pullup is properly enabled.

stronnag, 2019-08-18:
https://github.com/iNavFlight/inav/blob/956a3f5ed9aa634a7930a5e4229119f0773a2dde/src/main/target/common.h#L28

KilianE, 2019-08-19:
That means it is already activated, or do I have to activate it?

fiam, 2019-08-19:
@KilianE Internal pull-ups won't work for i2c in most cases. They're in the 30k-40k range, while you need something in the 1k-2.2k.

rrk1, 2019-08-19:
fiam, good point!

This a confusing topic, which generally in need for better documentation, because proper pull-up is critical.

The source of confusion is that sometimes I2C is terminated with resistors on the controller board (ones with I2C MPU6050 or baro on-board), or sometimes you get just two bare GPIO ports which are in need for external resistors. Some external boards like popular MS5611 baro module has its own 2.2K resistors. BN880 board, if I remember right, has pull-up on I2C, but too weak (around 10K)

Too much pull up is bad, too. 

Personally, I am always checking I2C on my new builds with oscilloscope for healthy waveform. It may be a good idea to anyone to get a scope and learn how to use it. (<=$100 those days for a scope that already is not a toy, do not get distracted by DSO201-203 crap and likes, get at least a cheap Hantek 2C72/JDS and up)

Another simple hardware test for checking I2C pull-up on already assembled system is to measure a resistance between SDA and SCL wires with a mutimeter, power off of course. It shall show about 4.4KOhm if proper resistors are present somewhere on the bus.

KilianE, 2019-08-19:
Is OK. The problem is that I am not at home at the moment. At home I have a good oscilloscope, every resistor value more then ten times and stuff like this... So it wouldnt be a problem, if I would be at home. But I got the point. When the internal pullup resistors are to high, then I dont want to use them. Otherwise It could be that I will loose signal, but then my quad will not return home... That would be a big problem... Thank you for supporting me, I will close it. 
Thank you very much that you are all carrying for this great software!!!
Have a good day!

closed 2019-08-19 by KilianE