iNavFlight/inav #5024: spektrum satellite bind problem with inav 2.2.1

ahmetsagir, 2019-08-18:
Hello there;

I use INAV 2.2.1 on OMNIBUS F4 Pro FCs. I can't connect the spectrum satellite.

I'm writing the cli command - spectrum_sat_bind = 9 and save and restart fc, but the satellite is not flashing. What could be the problem?

I tried it with betaflight. The result is successful. Satellite is working with betaflight but is not working with inav.

stronnag, 2019-08-18:
Which port are you using?

ahmetsagir, 2019-08-18:
I'm using UART1

stronnag, 2019-08-18:
Note also:
* You need to hard restart (i.e. powercycle)
* If the sat works in βF, it does not necessarily need to be re-bound for iNav
* Binding and use of spek sat works OK here

> I'm using UART1

Wrong. The bind pin pin is RX3 (as you're been told at  on RGC already).

ahmetsagir, 2019-08-18:
I don't fully understand. Can you explain some details?

stronnag, 2019-08-18:
https://github.com/iNavFlight/inav/blob/development/docs/Spektrum%20bind.md

and
https://github.com/iNavFlight/inav/blob/956a3f5ed9aa634a7930a5e4229119f0773a2dde/src/main/target/OMNIBUSF4/target.h#L273

ahmetsagir, 2019-08-18:
how can i change bind pin?

stronnag, 2019-08-18:
Move the RX cable from UART1 to UART3 for the bind, then move it back again :)

ahmetsagir, 2019-08-18:
I use dsmx satellite port in Fc. Because I have a gps and compass module and the compass module works on Uart3

I canceled the GPS and the compass to try.
I activated serial rx on uart3 and restarted the board and applied the bind procedure. result failed

ahmetsagir, 2019-08-18:
I Connected SAT to my receiver and bind both together to my Tx than it was successful. The satellite connection appears, but there is no movement on the receiver tab. FC does not detect radio movements.

closed 2019-10-31 by stale[bot]