iNavFlight/inav #5025: Matek 405STD , UARTS and I2C

robohippy, 2019-08-18:
Matek 405STD , UARTS and I2C

I have connected the caddx fc cable to RX4 and TX4,
if i enable softserial on TX4 and TX2 RX2, these signal will be inverted. I have read from the documents that i can invert this in the CLI command. Would i be correct in this assumption?

The second point is that on the underneath of the Matek 405STD there are SDA and SDL pad, do i use these or RX3 and TX3. Is it possible to use the SDA and SCL on the underneath and USE UART 3 as normal ?  Could i re assign S7 on Matek405STD as I2C via Custom firmware ?

[INAVCD.pdf](https://github.com/iNavFlight/inav/files/3513135/INAVCD.pdf)

I have included a diagram of where im upto, at the moment i have not connected the SDA and SDL.

Many thanks roboippyv2

teckel12, 2019-08-18:
@robohippy The board has 4 UARTS 1,3,4,5 that are all uninverted. UART 2 is inverted for use with SBUS.  Why are you wanting to invert the UART? Nothing you mention uses inverted UARTs and UART 2 has an inverter already.

I believe you can use either UART 3 **OR** I2C (SDA & SDL pads). 

Maybe you could express what you're trying to do? Connect the Caddx? Connect an FrSky SBUS receiver? Maybe something else? Just a little confused about what you're trying to accomplish.

robohippy, 2019-08-18:
i am trying to use the caddx fc cable which uses a UART and use it on UART4 as i have no other free uarts
since i have to enable UART 2 and 4 as SoftSerial, im assuming the signals will be inverted to work with FrSky. Since the caddx fc cable uses a normal UART i would have tochange the way it deals with UART 4 via CLI ?

 I am using the FrSKy R-SXR with F.Port firmware, so SBUS and S.Port will be merged into 1 signal and sent via the R-SXR from its S.Port output to the pad labled SBUS on the FC. i have included a PDF of the setup. many thanks for the reply, im just starting to learn about drones.

teckel12, 2019-08-18:
@robohippy Looking at your diagram it looks mostly correct. I don't understand why you have two receivers XM+ and X-SXR. And you have the S.Port (telemetry) on the X-SXR going to the SBUS port on the flight controller.

But really, I'm missing what your trying to solve here, and the two receivers and their connection are even more confusing.

robohippy, 2019-08-18:
what i want to know is it ok to put the Caddx fc cable on TX4 ? im assuming i will have to change its signal type via CLI.

The two receivers are for redundancy, they were the two receivers recommended to pair by FrSky. S.Port after flashing the firmware to the R-SXR becomes F.Port which is bothe the SBUS signal and the S.Port signal into one protocol.

robohippy, 2019-08-18:
i am trying connecting the gps to uart1 , fport to Uart 2, Uart 3 is used by I2C, The caddx fc cable used by the caxx tarsier camera to uart 4, and the vtx smartaaudio to tx5.

teckel12, 2019-08-18:
The Caddx will work on any UART, 1-5 as long as it's the only thing on that UART.  And I assume when you say "change its signal via CLI" you believe it needs to be inverted?  First, you can't invert UARTS on a F4.  It requires hardware or a soft serial port to invert as the F4 doesn't have the ability to invert on it's own like the F3/F7.  Secondly, the ONLY thing that ever needs to be inverted is old FrSky receivers.  Many newer FrSky receivers either default to uninverted or have both inverted and uninverted ports.  Anyway, no idea why you think you need something set in CLI to invert the signal for the Caddx.  You're making this harder than it is.

There's only one inverted port on the Matek F405STD, the one labeled SBUS.  The TX2 port is uninverted.

I see what you're doing with the redundancy, is that even needed?  Seems going with a 900Mhz receiver would be a better option.  I've never had a receiver fail, let alone in the air.

It sounds like what you want to do will work.  GPS on UART1, FPort on TX2 (or SBUS depending on inversion), Caddx on UART4, VTX on UART5.

What are you using on the I2C bus?  Magnetometer?  I assume this is a fixed wing because you have a servo?  If so, there's no really a need for a magnetometer.

robohippy, 2019-08-18:
thank you for the reply, excuse my text, in new at this :) and not so good in text. you have explained very clearly and now i comprehend. we are using the redundancy as we were given the xm+ and it seemed a good idea to use it, even for an experiment.

Yes i agree a 900mhz ish (EU) would be better, we have looked into that but at the moment we just need to make it fly. Its taken us 3 months for all the parts to arrive and get to the point where i just about know what im doing lol.

On the I2C i am using the magnetometer from the Matek GPS unit and i think its the barometer bmp280.

This is for a Quad the servo is for making a tilt mechanism for the camera.

i did scan through the code for assigning pins to add another servo to S7 for another axis for camera. It looks possible, since s5 is LED strip in inav, atlthough it looks like i might have to make a custom firmware. Although i dont know how that would affect the configurator.

but for now thank you for your help it has been great.

teckel12, 2019-08-19:
@robohippy If it's a quad, adding the magnatometer is the right thing to do as it's needed for navigation.

INAV doesn't have pin reassignment, so adding another servo would require a custom target. Looking at the INAV and Betaflight source for the Matek F405STD may give you the insight to do this. It won't require a y changes in the configurator. It will still work correctly to flash and change the settings in custom firmware.

robohippy, 2019-08-21:
hi i looked at the source and the pin and channel defintions for the servo6 firmware version.

i edited it to add a second servo onto pin s7, am i correct ?

edited inav/src/main/target/MATEKF405/target.c 

#include <stdbool.h>
#include <platform.h>
#include "drivers/io.h"
#include "drivers/pwm_mapping.h"
#include "drivers/timer.h"

const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM9, CH2, PA3,  TIM_USE_PPM,   0, 0), // PPM

#ifdef MATEKF405_SERVOS6
    DEF_TIM(TIM3, CH1, PC6,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_MOTOR, 0, 0), // S1
#else
    DEF_TIM(TIM3, CH1, PC6,  TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_MOTOR, 0, 0), // S1
#endif
    DEF_TIM(TIM8, CH2, PC7,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 1), // S2  UP(2,1)
    DEF_TIM(TIM8, CH3, PC8,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 1), // S3  UP(2,1)
    DEF_TIM(TIM8, CH4, PC9,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0), // S4  UP(2,1)

    DEF_TIM(TIM2, CH1, PA15, TIM_USE_MC_MOTOR | TIM_USE_LED                        , 0, 0), // S5  UP(1,7)

#ifdef MATEKF405_SERVOS6
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S6  UP(2,5)
#else
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0), // S6  UP(2,5)
#endif
    
#ifdef MATEKF405_SERVOS6
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S7  
#else
    DEF_TIM(TIM4, CH3, PB8,  TIM_USE_MC_MOTOR |                    TIM_USE_FW_SERVO, 0, 0), // S7  
#endif

    DEF_TIM(TIM5, CH3, PA2,  TIM_USE_PWM,                                            0, 0), // TX2

    DEF_TIM(TIM5, CH1, PA0,  TIM_USE_PWM,                                            0, 0),   // TX4 UP(1,6) D(1,2)!S1
    DEF_TIM(TIM5, CH2, PA1,  TIM_USE_PWM,                                            0, 0),   // RX4 UP(1,6) D(1,4)
};

const int timerHardwareCount = sizeof(timerHardware) / sizeof(timerHardware[0]);

robohippy, 2019-08-22:
https://github.com/iNavFlight/inav/blob/master/src/main/target/MATEKF405/target.h

i looked at the source code from the code it seems that Uart3 is used on the Matek 405osd only which seems to coresponf to the text in the docs.
https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20MatekF405.md

so seems in my case the matek 405std, which has the exposed pads on the underneath, the code assigns the i2c bus to the pads on the underneath. in the case of the 405osd it has the pins for uart 3 assigned to i2c

Matek 405 docs I2C

The F405-AIO, STD, CTR boards expose dedicated I2C pads. The F405-OSD does not expose I2C. For iNav there is a software I2C provision using the USART3 pads, as:

    SDA => RX3, SCL => TX3
    Do not assign any serial function to USART3
================================

// SLC clash with WS2812 LED
#ifdef MATEKF405OSD
    // OSD - no native I2C
    #define USE_I2C
    #define USE_I2C_DEVICE_EMULATED
    #define I2C_DEVICE_EMULATED_SHARES_UART3
    #define SOFT_I2C
    #define SOFT_I2C_SCL            PC10 //TX3 pad
    #define SOFT_I2C_SDA            PC11 //RX3 pad

    #define DEFAULT_I2C_BUS         BUS_I2C_EMULATED
#else
    // AIO / CTR / STD
    #define USE_I2C
    #define USE_I2C_DEVICE_1
    #define I2C1_SCL                PB6
    #define I2C1_SDA                PB7

    #define DEFAULT_I2C_BUS         BUS_I2C1
#endif

closed 2019-08-23 by robohippy

packocrayons, 2021-10-04:
Is there more involved on this than just flashing the F405OSD target and not enabling anything on UART3? Testing with an oscilloscope shows no activity (however are no longer pulled high) on RX3/TX3 with a magnetometer enabled on I2C_DEFAULT (which _is_ the software I2C, based on the header).

The SDA/SCL pads also show no activity on a scope.

Is a setup step missing somewhere in documentation, or has this bug cropped back up?

stronnag, 2021-10-04:
softI2C on MATEKF405OSD (sda=>rx3, scl=>tx3) works OK here.