iNavFlight/inav #5026: Compiling RunTime Error

RomanFoltyn, 2019-08-19:
Hallo,
in the future I need a special version of the F722-Wing.
To check the possibilities of compiling the standard version of the software (from Github) I installed the necessary components as recommended. I installed the latest GCC RUBY and CYGDRIVE versions.

**please help because despite installing earlier versions of programs error always occurs. I don't know what I'm doing wrong?
Unfortunately, RunTimeError appears during compilation:**

Best Regards 
Roman Foltyn  
_foltyn@foltyn.de_

Roman Foltyn@i5-W7-02 /cygdrive/c/dev/inav
$ make TARGET=MATEKF722SE
Traceback (most recent call last):
        3: from ./src/utils/build_stamp.rb:89:in `<main>'
        2: from ./src/utils/build_stamp.rb:44:in `stamp'
        1: from ./src/utils/build_stamp.rb:66:in `hash_config'
/cygdrive/c/dev/inav/src/utils/compiler.rb:92:in `run': Compiler error: (RuntimeError)
/cygdrive/c/Program Files (x86)/CodeSourcery/Sourcery_CodeBench_Lite_for_ARM_EABI/bin/arm-none-eabi-g++ -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-sp-d16 -fsingle-precision-constant -Wdouble-promotion -flto -fuse-linker-plugin -Os -I./src/main -I./src/main/target -I./lib/main/STM32F7/Drivers/STM32F7xx_HAL_Driver/Inc -I./lib/main/STM32F7/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I./lib/main/STM32F7/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc -I/Inc -I./lib/main/CMSIS/Core/Include -I./lib/main/STM32F7/Drivers/CMSIS/Device/ST/STM32F7xx/Include -I./src/main/vcp_hal -I./lib/main/FatFS -I./lib/main/MAVLink -I./src/main/target/MATEKF722SE -ggdb3 -DDEBUG -std=c++11 -Wall -Wextra -Wunsafe-loop-optimizations -Wdouble-promotion -Werror=switch -ffunction-sections -fdata-sections -fno-common -DUSE_HAL_DRIVER -DUSE_FULL_LL_DRIVER -DSTM32F722xx -DHSE_VALUE=8000000 -DARM_MATH_MATRIX_CHECK -DARM_MATH_ROUNDING -D__FPU_PRESENT=1 -DUNALIGNED_SUPPORT_DISABLE -DARM_MATH_CM7 -DFLASH_SIZE=512 -DHSE_VALUE=8000000 -DUSE_STDPERIPH_DRIVER -DMATEKF722SE -DSTM32F7 -DMATEKF722SE -D__FORKNAME__=inav -D__TARGET__=MATEKF722SE -D__REVISION__=956a3f5ed -I./obj/main/MATEKF722SE -dM -E -o ./obj/main/MATEKF722SE/stamp ./obj/main/MATEKF722SE/stamp.cpp
cc1plus.exe: error: bad value (cortex-m7) for -mcpu switch
cc1plus.exe: error: unrecognized command line option '-std=c++11'
cc1plus.exe: error: invalid floating point option: -mfpu=fpv5-sp-d16
make: *** [Makefile:316: obj/main/MATEKF722SE/build.stamp] Fehler 1

Roman Foltyn@i5-W7-02 /cygdrive/c/dev/inav

teckel12, 2019-08-19:
@RomanFoltyn The latest development version of INAV doesn't compile on Windows. Try the devopment branch, that should compile.

DzikuVx, 2019-09-06:
Please try alternative building methods for windows, cygwin always was at least problematic:

https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%20light.md
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%2010%20with%20Linux%20Subsystem.md
https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Docker.md

closed 2019-09-06 by DzikuVx