iNavFlight/inav #5030: no connection on save and reboot / disconnect / reconnect

bavariaaa, 2019-08-21:
## Current Behavior
When I disconnect and try to connect again via USB-MSP / BT-MSP / 3dr-MSP the port cannot be opened. It can only be opened once, then I have to restart the configurator / restart the FC.

## Steps to Reproduce
1. open configurator
2. connect via usb
3. disconnect
4. reconnect
5. not able to open port :/

Please tell me what additional data you need.
 [log](https://pastebin.com/xJyKL6XS)

---

# version
# INAV/MATEKF722SE 2.2.1 Jul  3 2019 / 22:44:40 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]

- MATEK F722 wing

teckel12, 2019-08-22:
@bavariaaa Define "disconnect" step 3.

Do you mean pressing the disconnect icon or disconnecting thr physical USB connection?

I know I've done I've disconnected with the button many times without a problem. May be the USB driver you're using.

bavariaaa, 2019-08-22:
by pressing the disconnect icon.
Everytime I do this, I have to either restart the configurator or restart the FC to get a connection again.

I'm using no special USB driver. I used Zadig to install latest winusb driver.
Noone having this issue?

Yours J

bavariaaa, 2019-08-22:
I accedentially closed this :( pls reopen.

stronnag, 2019-08-22:
No issues using a BT / 3DR bridge to connect / disconnect to an FC (MATEKF405) using either the configurator or mwp. Works every time, all the time. 

Looks like a host platform issue to me ... 

Perhaps you can further help in one of the support fora.

teckel12, 2019-08-22:
@bavariaaa I verified with a Matek F722SE and I was able to disconnect/reconnect without any issues. Try rebooting, it seems like a local platform/driver issue.

I2C errors are normal on boot. Contrary to common misconception, the I2C number is the quantity of errors, not an error number. So you got ~45 I2C errors at boot, which is a common quantity. As long as they stop counting up, you're good.

bavariaaa, 2019-08-22:
First of all thank you for your posts.
I just tried reconnecting on linux and it works fine.
I also tried another windows 10 machine and there it's like mentioned in my first post. No connection after disconnecting and reconnecting. Using Zadig and installing winusb driver is the right way, isn't it?

DzikuVx, 2019-08-23:
Actually you should install Stm vcp driver