iNavFlight/inav #5033: Fixedwing Trims in Adjustments

FPVZaphod, 2019-08-27:
## Current Behavior
Currently you need to use Autotrim or the Configurator to trim the plane. It is not possible to change the servo midpoint in flight. 

## Desired Behavior
As it's possible to adjust the Board alignment via in flight adjustments, it would be nice to be also able to change the trims for roll and pitch and yaw in flight. 

## Suggested Solution
3 more options in the Adjustment Tab:  Trim Roll, Trim Pitch and Trim Yaw 

## Who does this impact? Who is this for?
Airplane users, simplifies the tuning for Fixedwings.

Alexbombali, 2020-01-14:
Yes, please this. I find it very finicky to use auto trim and I would much rather map the trim tabs on my Tx to a channel and adjust the trims that way. This would make it a lot easier to make small corrections to the trims instead of doing it all over again with auto trim.

FPVZaphod, 2020-01-14:
Wondering why nobody else needs this.
How do you guys trim your plane, to make it fly perfect in manual mode? Autotrim nice, but there should be also a way for manual trim? I can change so many things in flight, but not something fundamental like the trims?
Please, Devs  - help :-)

closed 2024-02-02 by DzikuVx