iNavFlight/inav #5035: Strange Smart Port behavior on softserial using F411 Wing

michalpelka, 2019-08-28:
## Current Behavior
Hi,
I having hard time with Matek F411 and INAV 2.2.1. Board does not cooperate with Frsky receiver (X8R). I compared communication using F405 and F411. F411 works fine (Taranis reports from time to time sensor loss). With F405 it complains ~ 20 times per minute. I get one valid response from F411 on plot:
[image]

## Steps to Reproduce
1. Flash F411and wire F411 according to http://www.mateksys.com/?portfolio=f411-wing
2. Setup SBUS on UART1
3. Setup SOFSERIAL1 to SMARTPORT@AUTOBAUD
4. Suffer from "SENSOR LOST"

## Expected behavior
F411 should behave same as F405.

## Suggested solution(s)
 - Please someone confirm that setup works.
 - Is it possible to debug board with SWD?
 - Maybe it is bug in INAV target.

## Additional context

 [Dump from F411 - Not working](https://pastebin.com/A5ZLFadB) 
 [Dump from F405 - Working](https://pastebin.com/2rAmwY5h)
---

- FC Board name and vendor: Matek F411-Wing
- INAV version string: 
```
# version
# INAV/MATEKF411 2.2.1 Jul  3 2019 / 22:50:30 (a6d847482)
# GCC-8.2.1 20181213 (release) [gcc-8-branch revision 267074]
```

aMax52, 2019-09-03:
I had the same issue with ver2.1. and this controller The serial port seems not to be fast enough for all the data (running Inav lua). Since I changed my R9mm to fport firmware, it works.

DzikuVx, 2020-02-19:
SoftwareSerial should be avoided for anything faster than 19200 but sometimes even this is too fast. On top of that, F405 run on 168MHz while F411 on around 100MHz

closed 2020-02-19 by DzikuVx