iNavFlight/inav #5037: 2.2.1 Smartaudio don't work

yanzen1989, 2019-08-29:
1. FC Matek f405-ctr
2.Firmware INAV 2.2.1
3.VTX AKK DVR VTX

Problem with SMARTAUDIO.
If i flash two last firmwares smartaudio not work
If i downgrade firmware to 2.1.0 and load my config vtx smartaudio SA work stable

What's wrong?

https://pastebin.com/ESrdUDx6

Sorry for my speak, i'm Russian.

gregm412, 2019-09-01:
I have the same problem with the furious f35 flight controller and akk fx2 vtx.  Can anyone help here?

Mateyhv, 2019-09-01:
@yanzen1989 you mentioned smart audio but not the VTX. I guess it is an AKK VTX. If thats the case there is already open case somewhere for that issue for quite a long time already. Hope there comes some solution soon.

teckel12, 2019-09-01:
@issue-label-bot[bot] SA 2.1 is not yet fully supported with INAV.

leeph, 2019-09-01:
Any indication whether support will be provided in 2.2.2 or whatever the next release will be?

It's a shame to lose vTX control with latest SmartAudio devices.

teckel12, 2019-09-02:
@leeph I haven't got around to building a new model with a SA v2.1 VTx but it's on my list. No INAV 2.2.2 PRs resolve this at this point.

leeph, 2019-09-10:
Would it help if you were to be provided with a SA2.1-capable VTX for development, or is it more a time/resources issue?

SA2.1 code in Betaflight4.x is available so I would imagine adding support shouldn't be too onerous.

teckel12, 2019-09-10:
@leeph It's more of a time issue at this point. Also, I believe the SA2.1 stuff in Betaflight was part of a total rewrite of certain related things. So it's not as easy as just copying a few lines of code. Things need to be stripped out and/or refactored.

sirPerna, 2019-09-24:
If we can support this by supplying hardware with different SA versions for testing please let us know. 

TBS

leeph, 2019-09-24:
@sirperna it would be useful to know what SA changes exist in Unify Pro32HV firmware - the change log doesn't mention any but the fact is that SA is completely non-functioning in v1.08 but almost fully functioning in v1.12 save for incorrect power levels towards the top end.

sirPerna, 2019-09-24:
> @sirPerna it would be useful to know what SA changes exist in Unify Pro32HV firmware - the change log doesn't mention any but the fact is that SA is completely non-functioning in v1.08 but almost fully functioning in v1.12 beta save for incorrect power levels towards the top end.

The Unify Pro32 HV does report the supported power levels through SA. In early firmware versions there was a bug where the Unify reported too many power levels which made betalfight not accept it as the amount was bigger then the buffer they offer. We fixed this inside the Unify firmware a while ago. 

Also if INAV support the variable frame length it can thread SA2.1 VTX with the SA2.0 standard and they will still work.

putimir, 2020-01-06:
Hello all! May I ask if there is someting getting done on this request/issue? Can TBS make the hardware backward compatible with previous versions of SA?

Thanks

Mateyhv, 2020-01-13:
I would like to know if there is any solution available or if there is some workaround. Might be wrong as I am not following the issue but as far as I know this is solved in Beta

DzikuVx, 2020-05-27:
INAV 2.5 bring SA2.1 support

closed 2020-05-27 by DzikuVx