iNavFlight/inav #5041: About the Galileo satellite after its activation

ykyandy, 2019-08-31:
I activated the Galileo satellite in the flight control. Will the GPS satellite stop working if the Galileo satellite is enabled? Or gps satellites will work together?

By the way, is the accuracy of the Galileo satellite higher than GPS? Is it necessary to activate in Asia?

teckel12, 2019-08-31:
@ykyandy Galileo is just another 22 GPS satellites that the EU launched. When you turn on Galileo you get an additional 22 satellites to get a GPS fix. More satellites means possibly higher accuracy. When you turn on Galileo, it doesn't use them exclusively, it's in addition.

ykyandy, 2019-08-31:
> @ykyandy Galileo is just another 22 GPS satellites that the EU launched. When you turn on Galileo you get an additional 22 satellites to get a GPS fix. More satellites means possibly higher accuracy. When you turn on Galileo, it doesn't use them exclusively, it's in addition.

Thank you for the reply

stronnag, 2019-08-31:
Please see https://github.com/iNavFlight/inav/wiki/Ublox-3.01-firmware-and-Galileo which explains the allocation of GNSS and device channels by iNav.

closed 2019-09-02 by DzikuVx