iNavFlight/inav #5045: Glider always Armed, always RTH

fremebondo, 2019-09-04:
Hi, 
I've tried to figure out if there is a combination of settings capable of controlling a glider, dropped from the cargobay of an airplane, to fly back home without ANY pilot input, neither to arm or activate RTH mode.

I am planning to start dropping glider with RX on board and activate RTH with a second radio, but when the sequence is reliable, I wish to unequip the RX and left only a FC onboard the glider. In this case the glider would be unpowered while attached to cargo-plane: when dropped, a spring loaded contact closes and powers up the FC. During launch this spring loaded contact is mechanically kept opened by the plane fusellage.

ReCap: is there any way to have the FC immediately jump in ARMED-RTH mode without caring about RX inputs? Failsafe may be the answer, but I need to be in failsafe on power up. 

In case there is no such a capability out of the box I'll try to edit nav modes code to reach my goal. 

Many Thanks!
Fremebondo

fremebondo, 2019-09-04:
Also, Home position needs to be stored in some way to have this to work.

Alternatively Home could be saved on powerup, and RTH mode enabled by other conditions:
- digital input on FC board
- Z acceleration sign (if the glider is launched casized as attached to the plane, and naturally flips back leveled when released)

Pairan, 2019-09-04:
I don't know but maybe my thoughts are usefull in some way!

How about setting a waypoint on the SD and power it in waypoint mode!? ... but how would it get there if it has no satelite fix!?

I guess it needs to be powered up in order to find home properly (homepoint must be set, gps fix).

Mateyhv, 2019-09-05:
Possibly you can use launch mode modified to be activated on gliders release and the glider should remain armed the whole ferry flight. For home position I am not sure if after locking a home position and then setting the reset home position parameter to none instead of on first arm or on every arm, the FC will remember the original fixed position on each reboot. You may test it by yourself.

fremebondo, 2019-09-23:
@Pairan, you are right about GPS fix and power on! The glider has definitely to be powered when still on the ground, then I need to find a way to activate RTH or Waypoint when released from the vector plane. 
Maybe I could redefine a ServoOutput to be used as DigitalInput and close it with a spring loaded contact, closed at glider's release.

@Mateyhv idea of using launch mode: relying on accelerometer data to trigger RTH may be misleading, as accelerometer senses a lot of acceleration during takeoff and climb, when the glider is attached to the vector. Maybe I could try to sense a free fall, but also an uncontrolled momentary dive of the vector plane can easily reach -9,8m/s^2 and trigger RTH on the glider when still attached. Maybe I could sense if the freefall is continued for 5 seconds or more... but I think this sequence is a little unreliable.

closed 2019-12-06 by stale[bot]